GabMus/razerCommander #1: Redesign

julianrichen, 2016-07-10:
Hello,
I don't own a Razer device but was impressed by your work, thank you for making a re-implantation of something that will probably never be ported to Linux.

I wanted to provide some suggestions on improving the ui.
[image]

You could also make the Breath Mode section a combo box/pop-over to save more room for more features later on.

You can view the original [svg here](http://dl.richen.io/gnome/razerCommander/razerCommander.svg)

GabMus, 2016-07-10:
It's done! I'm pushing the changes in like 1 minute

closed 2016-07-10 by GabMus

julianrichen, 2016-07-10:
Damn that was fast, glad it helped :)

My only other suggestion would be to use a consistent padding. I believe GNOME uses 7px (or 10px) for almost everything with extra padding being multiples of 7, 14px, 21px, etc...

Oh and I just realized your the owner of TechPill, I actually found this app from your channel but I didn't connect the two. Thanks for promoting Linux.