GabMus/razerCommander #7: Error when trying to use razerCommander

Kurolox, 2016-08-25:
Installed from the AUR. The razer-driver was installed manually and tested.

This is the output of both razercommander and python /route/to/main.py:

`** (main.py:25443): WARNING **: Couldn't register with accessibility bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Traceback (most recent call last):
  File "/usr/lib/python3.5/site-packages/dbus/bus.py", line 175, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/usr/lib/python3.5/site-packages/dbus/bus.py", line 361, in get_name_owner
    's', (bus_name,), **keywords)
  File "/usr/lib/python3.5/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.razer': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.5/site-packages/razer/client/__init__.py", line 15, in __init__
    self._dbus = session_bus.get_object("org.razer", "/org/razer")
  File "/usr/lib/python3.5/site-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3.5/site-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python3.5/site-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3.5/site-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python3.5/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.razer was not provided by any .service files

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/share/razercommander/main.py", line 7, in <module>
    import device
  File "/usr/share/razercommander/device.py", line 10, in <module>
    device_manager = rclient.DeviceManager()
  File "/usr/lib/python3.5/site-packages/razer/client/__init__.py", line 17, in __init__
    raise DaemonNotFound("Could not connect to daemon")
razer.client.DaemonNotFound: Could not connect to daemon
`

GabMus, 2016-08-25:
Apparently the daemon isn't running, try running `razer-service` on your
terminal and try again

GabMus, 2016-08-25:
And I just noticed you said that you installed the driver "manually". Do you mean using make? If so, try installing using the provided PKGBUILDs in terrycain's repo

kusayuzayushko, 2016-08-28:
Same problem here. razer-driver-dkms, razer-daemon and python-razer installed using provided PKGBUILDs. 
razercommander faild with
```                                                                                                                  
Traceback (most recent call last):
  File "/usr/lib/python3.5/site-packages/dbus/bus.py", line 175, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/usr/lib/python3.5/site-packages/dbus/bus.py", line 361, in get_name_owner
    's', (bus_name,), **keywords)
  File "/usr/lib/python3.5/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.razer': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.5/site-packages/razer/client/__init__.py", line 15, in __init__
    self._dbus = session_bus.get_object("org.razer", "/org/razer")
  File "/usr/lib/python3.5/site-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3.5/site-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python3.5/site-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3.5/site-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python3.5/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.razer was not provided by any .service files

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/share/razercommander/main.py", line 7, in <module>
    import device
  File "/usr/share/razercommander/device.py", line 10, in <module>
    device_manager = rclient.DeviceManager()
  File "/usr/lib/python3.5/site-packages/razer/client/__init__.py", line 17, in __init__
    raise DaemonNotFound("Could not connect to daemon")
razer.client.DaemonNotFound: Could not connect to daemon
```
razer-service faled with
```                                                                                      
Traceback (most recent call last):
  File "/usr/bin/razer-service", line 6, in <module>
    from razer_daemon.daemon import daemonize
  File "/usr/lib/python3.5/site-packages/razer_daemon/daemon.py", line 23, in <module>
    import razer_daemon.hardware
  File "/usr/lib/python3.5/site-packages/razer_daemon/hardware/__init__.py", line 7, in <module>
    from razer_daemon.hardware.mouse import RazerMambaChromaWireless
  File "/usr/lib/python3.5/site-packages/razer_daemon/hardware/mouse.py", line 5, in <module>
    from razer_daemon.misc.battery_notifier import BatteryManager
  File "/usr/lib/python3.5/site-packages/razer_daemon/misc/battery_notifier.py", line 8, in <module>
    import notify2
ImportError: No module named 'notify2'
```
Iven thought that's not your problem, I'm just going to let this here.

GabMus, 2016-08-28:
EASY PEASY: just install `python-notify2` from AUR

ZeroKelvinKeyboard, 2016-08-29:
Not sure about AUR but on Ubuntu you have to install `python3-notify2`, which is a separate package to `python-notify2`.

GabMus, 2016-08-29:
In arch packaging python is python3 and python2 is python2, so the right
package to install is python-notify2 from AUR

GabMus, 2016-08-29:
@kusayuzayushko please report back and tell me if your issue is fixed so I can close it

GabMus, 2016-08-31:
Closing for now, assuming it's solves. If it's not, reopen it

closed 2016-08-31 by GabMus