GabMus/razerCommander #9: Translation?

geordieF, 2016-08-31:
Some of the translations are just weird. It would be nice to also have community translations, but glade may be a problem here. I can do a bit of basic Czech and English. BTW you should change "Pipette" to "eyedropper"

GabMus, 2016-08-31:
Alright cool. I'll try to implement translations ASAP. If you can roughly
translate in czech (give me couples of <english word>: <czech
translation>)  it'd be nice.
As for english, I thought it was all correct english. Maybe some italian
slipped in the process? Please just tell me if you see any around. About
pipette, I thought it was english, gonna fix that as well 😀

geordieF, 2016-09-01:
Pipette _is_ an English word, just not the right one here. Had a look into it and Glade actually has support for translations, so later today i'll have a run-through to see if anything else is wrong.

geordieF, 2016-09-01:
On a different note, instead of clicking a key, we could just make is so you just press that key to toggle select it

GabMus, 2016-09-01:
> instead of clicking a key, we could just make is so you just press that key to toggle select it
Sorry, couls you be more clear?

GabMus, 2016-11-21:
Closed for inactivity

closed 2016-11-21 by GabMus