GabMus/razerCommander #12: Error

xeq, 2016-09-07:
I'm on the latest Ubuntu, when i run razerCommander in my terminal here's what it says
`root@secret:~# razercommander
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 175, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 361, in get_name_owner
    's', (bus_name,), **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.razer': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 15, in __init__
    self._dbus = session_bus.get_object("org.razer", "/org/razer")
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.razer was not provided by any .service files

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/share/razercommander/main.py", line 7, in <module>
    import device
  File "/usr/share/razercommander/device.py", line 10, in <module>
    device_manager = rclient.DeviceManager()
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 17, in __init__
    raise DaemonNotFound("Could not connect to daemon")
razer.client.DaemonNotFound: Could not connect to daemon
`

GabMus, 2016-09-07:
The daemon is not running. How did you install it?

01generator, 2016-09-08:
I am facing the same issue

GabMus, 2016-09-08:
@01generator possibly the driver isn't binding the devices. What distro are you running? where did you install the driver?

01generator, 2016-09-08:
I am using kubuntu 16.04 and I installed the drivers through ppa

GabMus, 2016-09-08:
Give me the log in the position `~/.razer-service/logs/razer.log`
what's the output of `ls -lH /sys/bus/hid/drivers/razerkbd/0003:1532*`?
are you part of the plugdev group? (give me the output of `cat /etc/group | grep plugdev`)

xeq, 2016-09-08:
fixed, it's compatibile only with the older one

GabMus, 2016-09-08:
Sorry could you be more clear? What works only with the older what?

01generator, 2016-09-09:
Hello,

Well in ~/.razer-service/logs/razer.log there are few warnings.
And all of them are like ```Could not find screensaver DBus``` and after this warning useually follows ```Could not find screensaver DBus```

There are also few DEBUG warnings ```Could not find screensaver DBus```

Here is a part of the log

```
2016-09-08 22:51:07 | razer                          | INFO     | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2016-09-08 22:51:07 | razer                          | INFO     | Adding razer.devices.disableTurnOffOnScreensaver method to DBus
2016-09-08 22:51:07 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2016-09-08 22:51:07 | razer                          | INFO     | Adding razer.daemon.stop method to DBus
2016-09-08 22:51:07 | razer                          | INFO     | Serving DBus
2016-09-08 22:51:09 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-08 22:51:09 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-08 22:51:09 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-08 22:51:09 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-08 22:51:09 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-08 22:51:09 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-08 22:51:09 | razer.screensaver              | WARNING  | Giving up on screensaver
2016-09-08 23:07:39 | razer                          | DEBUG    | DBus called get_serial_list
2016-09-08 23:07:51 | razer                          | DEBUG    | DBus called get_serial_list
2016-09-08 23:11:10 | razer                          | DEBUG    | DBus called get_serial_list
2016-09-08 23:11:25 | razer                          | DEBUG    | DBus called get_serial_list
2016-09-08 23:25:33 | razer                          | INFO     | Stopping daemon.
2016-09-09 10:03:39 | razer                          | INFO     | Initialising Daemon. Pid: 3310
2016-09-09 10:03:39 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-09-09 10:03:39 | razer                          | INFO     | Adding razer.devices.getDevices method to DBus
2016-09-09 10:03:39 | razer                          | INFO     | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2016-09-09 10:03:39 | razer                          | INFO     | Adding razer.devices.disableTurnOffOnScreensaver method to DBus
2016-09-09 10:03:39 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2016-09-09 10:03:39 | razer                          | INFO     | Adding razer.daemon.stop method to DBus
2016-09-09 10:03:39 | razer                          | INFO     | Serving DBus
2016-09-09 10:03:41 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-09 10:03:41 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-09 10:03:41 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-09 10:03:41 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-09 10:03:41 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-09 10:03:41 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-09 10:03:41 | razer.screensaver              | WARNING  | Giving up on screensaver
```

Now as far as the ls -lH /sys/bus/hid/drivers/razerkbd/0003:1532*
Returns No such file or directory O.o

```cat /etc/group | grep plugdev``` returns ```plugdev:x:46:nikolas```

GabMus, 2016-09-09:
Seems like the devices aren't binding to the driver correctly. Contact me
on telegram @gabmus at 4 pm GMT+1 (or later) and I can help you sort this
out :)

xeq, 2016-09-09:
the older daemon (the new one compiled from src doesn't work)

GabMus, 2016-09-09:
You need to run the latest daemon, we don't officially support older
versions

DasHammett, 2016-09-17:
I have the same error as the OP, with latest razer-driver installed manually on a Gentoo machine with kernel 4.7.3
It works fine if I execute it as root though...

My user is in the plugdev group

terrycain, 2016-09-17:
@DasHammett is the daemon running? What devices do you have? Are they supported by the daemon? And send the daemon log file (put it on pastebin)

DasHammett, 2016-09-18:
Hi, 

Yes the daemon is running (otherwise I guess it would not run as root either). I have a razer Blackwidow Ultimante 2016, so should be supported by both the driver and daemon.

```
2016-09-18 11:58:32 | razer                          | INFO     | Initialising Daemon. Pid: 3999
2016-09-18 11:58:32 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-09-18 11:58:32 | razer                          | INFO     | Found device.0: 0003:1532:0214.0005
2016-09-18 11:58:32 | razer.device0                  | INFO     | Initialising device.0 RazerBlackWidow2016
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding getSerial method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.misc.suspendDevice method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.misc.resumeDevice method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.misc.getVidPid method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.misc.getFirmware method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.misc.getMatrixDimensions method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.misc.hasMatrix method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceName method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceType method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.getBrightness method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.setBrightness method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setWave method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setStatic method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setReactive method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setNone method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathRandom method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathSingle method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathDual method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setCustom method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setKeyRow method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.macro.enableMacroKeys method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.led.gamemode.getGameMode method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.led.gamemode.setGameMode method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.led.macromode.getMacroMode method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.led.macromode.setMacroMode method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.led.macromode.getMacroEffect method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.led.macromode.setMacroEffect method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.macro.getMacros method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.macro.deleteMacro method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.macro.addMacro method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.lighting.bw2016.setStarlight method to DBus
2016-09-18 11:58:32 | razer.device0                  | DEBUG    | Adding razer.device.lighting.custom.setRipple method to DBus
2016-09-18 11:58:32 | razer.device0.keymanager       | DEBUG    | Starting KeyWatcher
2016-09-18 11:58:32 | razer                          | INFO     | Adding razer.devices.getDevices method to DBus
2016-09-18 11:58:32 | razer                          | INFO     | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2016-09-18 11:58:32 | razer                          | INFO     | Adding razer.devices.disableTurnOffOnScreensaver method to DBus
2016-09-18 11:58:32 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2016-09-18 11:58:32 | razer                          | INFO     | Adding razer.daemon.stop method to DBus
2016-09-18 11:58:32 | razer                          | INFO     | Serving DBus
2016-09-18 11:58:34 | razer.screensaver              | INFO     | Initialising DBus screensaver object
[... cut ...] Bus
2016-09-18 11:58:34 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-18 11:58:34 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-18 11:58:34 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-18 11:58:34 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-18 11:58:34 | razer.screensaver              | WARNING  | Giving up on screensaver
```

terrycain, 2016-09-18:
Hmm most odd, should just work.

DasHammett, 2016-12-19:
Hi, 

Any news on this?

terrycain, 2016-12-19:
Whats the issue again?

DasHammett, 2016-12-19:
Razercommander does not start with error message on the original post. As user razercommander cannot connect to the daemon, but as root, I have not problem.

I can provide again all logs you require.

Thank you

terrycain, 2016-12-19:
Your saying as root you can use it fine?

terrycain, 2016-12-19:
Also run `ps -aux | grep razer-service`

GabMus, 2017-01-26:
Any news on this?

GabMus, 2017-02-08:
Closed for inactivity

closed 2017-02-08 by GabMus