GabMus/razerCommander #14: 'no devices connected' when launching main.py

ShannonVanWagner, 2016-09-29:
Hi, I've installed the razer keyboard driver from: ppa:terrz/razerutils at https://github.com/terrycain/razer-drivers and the driver appears to load:

When I launch main.py:
~/razercommander$ ./main.py 
no devices
no devices
#Clicked 'Apply' button and get:
Traceback (most recent call last):
  File "./main.py", line 547, in on_universalApplyButton_clicked
    myrazerkb.setBrightness(int(newVal))
AttributeError: 'NoneType' object has no attribute 'setBrightness'

$ lsmod|grep razer
razerkbd               32768  0
hid                   118784  4 i2c_hid,hid_generic,razerkbd,usbhid

$ ps -ef |grep razer
guy+  2016  1916  0 22:25 ?        00:00:01 razerdaemon

$ modinfo razerkbd
filename:       /lib/modules/4.4.0-38-generic/updates/dkms/razerkbd.ko
license:        GPL v2
description:    Razer Keyboard Device Driver
author:         Terry Cain <terry@terrys-home.co.uk>
srcversion:     E3B82EF2D6590930D007CA6
alias:          hid:b0003g*v00001532p00000209
alias:          hid:b0003g*v00001532p00000216
alias:          hid:b0003g*v00001532p00000203
alias:          hid:b0003g*v00001532p00000208
alias:          hid:b0003g*v00001532p00000205
alias:          hid:b0003g*v00001532p00000214
alias:          hid:b0003g*v00001532p0000011A
alias:          hid:b0003g*v00001532p0000010D
alias:          hid:b0003g*v00001532p0000011B
depends:        hid
vermagic:       4.4.0-38-generic SMP mod_unload modversions 

$ grep razer /lib/udev/rules.d/99-razer.rules
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="011b", ENV{is_razer_kbd}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="010d", ENV{is_razer_kbd}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="011a", ENV{is_razer_kbd}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0203", ENV{is_razer_kbd}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0205", ENV{is_razer_kbd}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0208", ENV{is_razer_kbd}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0209", ENV{is_razer_kbd}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0214", ENV{is_razer_kbd}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0216", ENV{is_razer_kbd}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0042", ENV{is_razer_mouse}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0044", ENV{is_razer_mouse}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0045", ENV{is_razer_mouse}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0046", ENV{is_razer_mouse}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0048", ENV{is_razer_mouse}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="002f", ENV{is_razer_mouse}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0c00", ENV{is_razer_firefly}="yes"
SUBSYSTEM=="hid", DRIVER=="hid-generic", ENV{is_razer_kbd}=="yes", RUN+="razer_mount razerkbd $kernel"
SUBSYSTEM=="hid", DRIVER=="hid-generic", ENV{is_razer_mouse}=="yes", ATTRS{bInterfaceNumber}=="00", RUN+="razer_mount razermouse $kernel"
SUBSYSTEM=="hid", DRIVER=="hid-generic", ENV{is_razer_firefly}=="yes", ATTRS{bInterfaceNumber}=="00", RUN+="razer_mount razerfirefly $kernel"

$ grep razer /var/log/kern.log
Sep 27 21:54:54 box kernel: [    8.441699] razerkbd: module verification failed: signature and/or required key missing - tainting kernel
Sep 27 21:54:54 box kernel: [    9.443990] razerkbd 0003:1532:0216.0002: input,hidraw0: USB HID v1.11 Keyboard [Razer Razer BlackWidow X Chroma] on usb-0000:00:14.0-1/input1
Sep 27 21:54:54 box kernel: [    9.444047] razerkbd 0003:1532:0216.0003: input,hidraw1: USB HID v1.11 Mouse [Razer Razer BlackWidow X Chroma] on usb-0000:00:14.0-1/input2
Sep 27 21:54:54 box kernel: [    9.500108] razerkbd 0003:1532:0216.0001: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer BlackWidow X Chroma] on usb-0000:00:14.0-1/input0
Sep 27 21:57:06 box kernel: [  145.185487] razerkbd 0003:1532:0216.0009: input,hidraw0: USB HID v1.11 Keyboard [Razer Razer BlackWidow X Chroma] on usb-0000:00:14.0-1/input0
Sep 27 21:57:06 box kernel: [  145.241389] razerkbd 0003:1532:0216.000A: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer BlackWidow X Chroma] on usb-0000:00:14.0-1/input1
Sep 27 21:57:06 box kernel: [  145.297383] razerkbd 0003:1532:0216.000B: input,hidraw2: USB HID v1.11 Mouse [Razer Razer BlackWidow X Chroma] on usb-0000:00:14.0-1/input2
Sep 27 22:11:16 box kernel: [    9.174666] razerkbd: module verification failed: signature and/or required key missing - tainting kernel
Sep 27 22:11:16 box kernel: [    9.527264] razerkbd 0003:1532:0216.0003: input,hidraw0: USB HID v1.11 Mouse [Razer Razer BlackWidow X Chroma] on usb-0000:00:14.0-1/input2
Sep 27 22:11:16 box kernel: [    9.583236] razerkbd 0003:1532:0216.0002: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer BlackWidow X Chroma] on usb-0000:00:14.0-1/input1
Sep 27 22:11:16 box kernel: [    9.639310] razerkbd 0003:1532:0216.0001: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer BlackWidow X Chroma] on usb-0000:00:14.0-1/input0
[... 2 lines cut ...]
Sep 27 22:26:59 box kernel: [  962.011676] razerkbd 0003:1532:0216.0009: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer BlackWidow X Chroma] on usb-0000:00:14.0-1/input0
Sep 28 09:07:19 box kernel: [    8.823732] razerkbd: module verification failed: signature and/or required key missing - tainting kernel
Sep 28 09:07:19 box kernel: [    9.943180] razerkbd 0003:1532:0216.0003: input,hidraw0: USB HID v1.11 Mouse [Razer Razer BlackWidow X Chroma] on usb-0000:00:14.0-1/input2
Sep 28 09:07:19 box kernel: [    9.998891] razerkbd 0003:1532:0216.0002: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer BlackWidow X Chroma] on usb-0000:00:14.0-1/input1
Sep 28 09:07:19 box kernel: [   10.058973] razerkbd 0003:1532:0216.0001: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer BlackWidow X Chroma] on usb-0000:00:14.0-1/input0
Sep 28 20:15:21 box kernel: [    8.799924] razerkbd: module verification failed: signature and/or required key missing - tainting kernel
Sep 28 20:15:21 box kernel: [    9.504786] razerkbd 0003:1532:0216.0003: input,hidraw0: USB HID v1.11 Mouse [Razer Razer BlackWidow X Chroma] on usb-0000:00:1a.0-1.2/input2
Sep 28 20:15:21 box kernel: [    9.560471] razerkbd 0003:1532:0216.0002: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer BlackWidow X Chroma] on usb-0000:00:1a.0-1.2/input1
Sep 28 20:15:21 box kernel: [    9.620575] razerkbd 0003:1532:0216.0001: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer BlackWidow X Chroma] on usb-0000:00:1a.0-1.2/input0
Sep 28 20:23:14 box kernel: [  494.029010] razerkbd 0003:1532:0216.0004: input,hidraw0: USB HID v1.11 Keyboard [Razer Razer BlackWidow X Chroma] on usb-0000:00:1a.0-1.2/input0
Sep 28 20:23:14 box kernel: [  494.084825] razerkbd 0003:1532:0216.0005: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer BlackWidow X Chroma] on usb-0000:00:1a.0-1.2/input1
Sep 28 20:23:14 box kernel: [  494.140786] razerkbd 0003:1532:0216.0006: input,hidraw2: USB HID v1.11 Mouse [Razer Razer BlackWidow X Chroma] on usb-0000:00:1a.0-1.2/input2
Sep 28 22:24:54 box kernel: [    9.215919] razerkbd: module verification failed: signature and/or required key missing - tainting kernel
Sep 28 22:24:54 box kernel: [    9.907625] razerkbd 0003:1532:0216.0003: input,hidraw0: USB HID v1.11 Mouse [Razer Razer BlackWidow X Chroma] on usb-0000:00:1a.0-1.2/input2
[... cut ...]
Sep 28 22:28:32 box kernel: [  239.319819] razerkbd 0003:1532:0216.0008: input,hidraw0: USB HID v1.11 Keyboard [Razer Razer BlackWidow X Chroma] on usb-0000:00:1a.0-1.2/input0
Sep 28 22:28:32 box kernel: [  239.375702] razerkbd 0003:1532:0216.0009: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer BlackWidow X Chroma] on usb-0000:00:1a.0-1.2/input1
Sep 28 22:28:33 box kernel: [  239.432095] razerkbd 0003:1532:0216.000A: input,hidraw2: USB HID v1.11 Mouse [Razer Razer BlackWidow X Chroma] on usb-0000:00:1a.0-1.2/input2

[image]

Also tried rebooting, replugging, etc.

Any suggestions for how to get this working? Thanks!

GabMus, 2016-09-29:
Try version 1.0.1 (the previous version) and tell me if you see the window
at all

ShannonVanWagner, 2016-10-01:
Hi, so I tried 1.0.1 and it works awesome! In fact, 1.0.3 works now as well!

Something I had done is ran sudo apt-get upgrade just before trying your suggestion, it updated razer-daemon and related dkms. Now that I think about it, I should have tested your suggestion before updating those, but oh well, it works now!

Consider this issue resolved from my perspective. 
Thanks for this awesome software!

Related update version info:
$ apt-cache show  razer-daemon razer-kernel-modules-dkms
Package: razer-daemon
Source: razer
Priority: optional
Section: misc
Installed-Size: 244
Maintainer: Terry Cain <terry@terrys-home.co.uk>
Architecture: amd64
Version: 1.0.13-0ubuntu1
Recommends: python3-razer
Depends: razer-kernel-modules-dkms, python3, python3-dbus, python3-gi, python3-setproctitle, python3-notify2, xautomation, xdotool
Filename: pool/main/r/razer/razer-daemon_1.0.13-0ubuntu1_amd64.deb
Size: 35384
MD5sum: 1216cf6f4c67950dd9548f3902b87125
SHA1: 708267baed87b62d799418c97e928cf39d010d24
SHA256: 8361b565d472d954d51c97e8748d7dae31c8e64486666d03069864531ae286bb
Description-en: Razer Device Daemon
 A user-space daemon used to interface with the driver.
 It auto-starts in the XDG directory and handles syncing
 across devices.
Description-md5: 23f4ffdc1a25ec237b8f92399a66dcc5

Package: razer-kernel-modules-dkms
Source: razer
Priority: optional
Section: kernel
Installed-Size: 229
Maintainer: Terry Cain <terry@terrys-home.co.uk>
Architecture: amd64
Version: 1.0.13-0ubuntu1
Depends: dkms, udev
Filename: pool/main/r/razer/razer-kernel-modules-dkms_1.0.13-0ubuntu1_amd64.deb
Size: 24758
MD5sum: a3aacce7e761d3c9090b681a7341a666
SHA1: 856cfa2069a7c4635ec5d7fd4730ca4922f7b5c9
SHA256: e165c5d995c19481bf2a33c8836b83bdf4eaac8740c2582b58c2bd791e0cc47b
Description-en: Razer Peripheral Drivers
 A keyboard driver, a mouse driver and a mousemat driver
 for Razer devices, plus support for Razer Blade devices.
Description-md5: 413d0dbbefe5191defeb482915992704

closed 2016-10-01 by ShannonVanWagner