GabMus/razerCommander #18: I have issue running application

fr1sk, 2016-12-28:
Hi, i installed razer drivers, and cloned your repo. When im trying to run main.py i got this error

`Unable to init server: Could not connect: Connection refused
Unable to init server: Could not connect: Connection refused
Traceback (most recent call last):
  File "main.py", line 7, in <module>
    import device
  File "/home/fr1sk/razerCommander/razerCommander/device.py", line 10, in <module>
    device_manager = rclient.DeviceManager()
  File "/usr/lib/python3.5/site-packages/razer/client/__init__.py", line 15, in __init__
    session_bus = _dbus.SessionBus()
  File "/usr/lib64/python3.5/site-packages/dbus/_dbus.py", line 211, in __new__
    mainloop=mainloop)
  File "/usr/lib64/python3.5/site-packages/dbus/_dbus.py", line 100, in __new__
    bus = BusConnection.__new__(subclass, bus_type, mainloop=mainloop)
  File "/usr/lib64/python3.5/site-packages/dbus/bus.py", line 122, in __new__
    bus = cls._new_for_bus(address_or_type, mainloop=mainloop)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
`

When i run razer-service nothing happends but when i cat log form /home/fr1sk/.razer-service/logs it prints this:

`2016-12-28 01:01:39 | razer                          | INFO     | Initialising Daemon (v1.1.3). Pid: 2021
2016-12-28 01:01:39 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-12-28 01:01:39 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2016-12-28 01:01:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2016-12-28 01:01:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2016-12-28 01:01:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2016-12-28 01:01:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2016-12-28 01:01:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2016-12-28 01:01:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2016-12-28 01:01:39 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2016-12-28 01:01:39 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2016-12-28 01:01:39 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2016-12-28 01:01:39 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2016-12-28 01:01:39 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderChroma ------------------- (1532:0043)
2016-12-28 01:01:39 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2016-12-28 01:01:39 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
2016-12-28 01:01:39 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
2016-12-28 01:01:39 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2016-12-28 01:01:39 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2016-12-28 01:01:39 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
2016-12-28 01:01:39 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
2016-12-28 01:01:39 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0208)
2016-12-28 01:01:39 | razer                          | INFO     | Adding razer.devices.getDevices method to DBus
2016-12-28 01:01:39 | razer                          | INFO     | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2016-12-28 01:01:39 | razer                          | INFO     | Adding razer.devices.disableTurnOffOnScreensaver method to DBus
2016-12-28 01:01:39 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2016-12-28 01:01:39 | razer                          | INFO     | Adding razer.daemon.version method to DBus
2016-12-28 01:01:39 | razer                          | INFO     | Adding razer.daemon.stop method to DBus
2016-12-28 01:01:39 | razer                          | INFO     | Serving DBus
2016-12-28 01:01:41 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-12-28 01:01:41 | razer.screensaver              | ERROR    | Caught exception in run loop
Traceback (most recent call last):
  File "/usr/lib/python3.5/site-packages/razer_daemon/misc/screensaver_thread.py", line 123, in run
    screensaver_active = self._dbus_interface.GetActive()
  File "/usr/lib64/python3.5/site-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib64/python3.5/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib64/python3.5/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NotSupported: This method is not implemented
2016-12-28 01:20:13 | razer                          | INFO     | Stopping daemon on signal 15
2016-12-28 01:20:13 | razer.screensaver              | INFO     | Screensaver Thread finished
2016-12-28 01:21:04 | razer                          | INFO     | Initialising Daemon (v1.1.3). Pid: 1996
2016-12-28 01:21:04 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-12-28 01:21:06 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-12-28 01:21:06 | razer.screensaver              | ERROR    | Caught exception in run loop
Traceback (most recent call last):
[... 14 lines cut ...]
2016-12-28 01:21:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2016-12-28 01:21:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2016-12-28 01:21:06 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2016-12-28 01:21:06 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2016-12-28 01:21:06 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2016-12-28 01:21:06 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2016-12-28 01:21:06 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderChroma ------------------- (1532:0043)
2016-12-28 01:21:06 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2016-12-28 01:21:06 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
2016-12-28 01:21:06 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
2016-12-28 01:21:06 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2016-12-28 01:21:06 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
[... cut ...]
2016-12-28 01:21:06 | razer                          | INFO     | Adding razer.devices.disableTurnOffOnScreensaver method to DBus
2016-12-28 01:21:06 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2016-12-28 01:21:06 | razer                          | INFO     | Adding razer.daemon.version method to DBus
2016-12-28 01:21:06 | razer                          | INFO     | Adding razer.daemon.stop method to DBus
2016-12-28 01:21:06 | razer                          | INFO     | Serving DBus
`

BTW I am running fedora 25. Any suggestions?

GabMus, 2016-12-29:
Looks like a daemon problem, you should open an issue in the driver's repo.
Also, are you running a Wayland session? If so, please try logging into a
xorg session and report back.

GabMus, 2017-02-08:
Closing for inactivity

closed 2017-02-08 by GabMus