GabMus/razerCommander #19: Orbweaver support

mjtorn, 2017-01-04:
Hi!

Tested the packages linked in https://github.com/terrycain/razer-drivers/issues/22 - which make the Orbweaver visible to Polychromatic - but razerCommander does not seem to support it.

Hitting refresh does not make it visible.

Should this work?

Thanks!

GabMus, 2017-01-04:
Tartarus-like devices support is kinda weird right now in razercommander, the tartarus is somewhat supported, the orbweaver should work as well since it's basically the same thing, but I have to add a case for it. Will reach you out whenever possible

mjtorn, 2017-01-04:
Thanks a lot :) Just drop a line in this issue and I'll catch the email notification. Would be nice if #20 Just Worked(tm) because it'd make my end of the testing much easier, but if you require anything for it, don't hesitate to ask!

mjtorn, 2017-01-20:
Thanks for the work on #20 !

I upgraded my razer drivers to the Naga Hex preview .deb packages, this all works
(https://github.com/terrycain/razer-drivers/issues/149) and should include Orbweaver
too.

The way I tried razerCommander now was copying over the tarball of commit 726573a3893413ac05a3c8f3350aaec01376c006 and launching `./main.py`
but, alas, it claims no devices :/

GabMus, 2017-02-08:
Hey there, sorry for the long time it took me, but I finally have implemented a brand new macro system, that should make supporting new macro devices easy peasy.

I'd really need you to test this out for me, since my only device, a BW chroma tournament, doesn't have any macro functionality.

The new macro system is currently in the `macro_ng` branch, so if you want to clone it and test it out, that'd be great!

Also, if you can reach me out on Telegram (`@gabmus`) it'd be great, as I can help you test the new features, and eventually fix stuff on the fly.

GabMus, 2017-02-11:
@mjtorn you there?

mjtorn, 2017-02-11:
Sorry, been distracted by other things (like work and playing Richard & Alice under Crossover Games) but thanks to the weekend I got a moment to geek out on this. Too bad the moment didn't last very long, because commit df079acb235806b7196e5d7278778e83e300e94d only recognizes my Naga Hex mouse.

This is despite restarting the daemon.

BTW trying to change the settings, I get this

```
Traceback (most recent call last):
  File "./main.py", line 466, in on_universalApplyButton_clicked
    currentFX = fxListBox.get_selected_row().value
AttributeError: 'NoneType' object has no attribute 'value'
```

Kinda off-topic for this issue but is this related to the daemon or driver?

FWIW I'm currently running the snapshots I got in https://github.com/terrycain/razer-drivers/issues/149

Thanks!

GabMus, 2017-02-11:
Alright, there is some design inconsistency still. Differently from the fx section, the macro section doesn't require you to apply manually. In fact, once you assign a macro, it is registered without the press of the apply button. I should really hide that button in the macro section, thanks for the help!
Now, I'm not sure why razerCommander still doesn't recognize the orbweaver, so if you could please run this script (http://termbin.com/d717) and give me the output I could try to sort it out!
Again, if you contact me on telegram, we can sort it out together a lot quicklier

mjtorn, 2017-02-11:
```
Razer Orbweaver orbweaver
Razer Naga Hex mouse
```

I'm not on Telegram so I'm uneasy about joining just for the occasional razer thing, sorry :/

This is kinda weird, though, that it's not found.

mjtorn, 2017-02-11:
Also whenever I hit refresh, I get an extra Naga Hex in the device list ;)

GabMus, 2017-02-11:
alright, try to pull and retest

mjtorn, 2017-02-11:
Sorry, no success.

GabMus, 2017-02-11:
the orbweaver still doesn't show up?

GabMus, 2017-02-11:
Could you please give me more details?

mjtorn, 2017-02-11:
It does not show up, refreshing duplicates the Naga Hex and your python snippet produces the same output. So it did not alter any behavior.

GabMus, 2017-02-11:
Are you sure you're in the macros_ng branch? could you please give me the output of `git branch`?
The code responsible for adding devices is the following, it coudn't get simpler

    device_manager = rclient.DeviceManager()
    devlist = []
    for device in device_manager.devices:
        devlist.append(device)

mjtorn, 2017-02-11:
You were right about the branch. I don't want to install Git and stuff on my Steam Machine, so I do git exports and copy tarballs over if no .deb is available. I pulled to the latest commit, e5038996b835a17e8705cb7167d4cfda0e1a4ea2, and got this

```
desktop@steamos:~/Downloads/razerCommander$ ./main.py 

(main.py:2945): Gtk-WARNING **: /usr/src/packages/BUILD/gtk+3.0-3.14.5/./gtk/gtkcontainer.c:1174: child property `position' of container class `GtkHeaderBar' is not writable

(main.py:2945): Gtk-WARNING **: /usr/src/packages/BUILD/gtk+3.0-3.14.5/./gtk/gtkcontainer.c:1174: child property `position' of container class `GtkHeaderBar' is not writable

(main.py:2945): Gtk-WARNING **: /usr/src/packages/BUILD/gtk+3.0-3.14.5/./gtk/gtkcontainer.c:1174: child property `position' of container class `GtkHeaderBar' is not writable

(main.py:2945): Gtk-WARNING **: /usr/src/packages/BUILD/gtk+3.0-3.14.5/./gtk/gtkcontainer.c:1174: child property `position' of container class `GtkHeaderBar' is not writable
Traceback (most recent call last):
  File "./main.py", line 74, in <module>
    refreshDevices()
  File "./main.py", line 70, in refreshDevices
    initDevices()
  File "./main.py", line 36, in initDevices
    newdev = device.Device(dev)
  File "/home/desktop/Downloads/razerCommander/device.py", line 30, in __init__
    self.device)
  File "/home/desktop/Downloads/razerCommander/macro_logic.py", line 73, in make_device
    n_device = MacroDevice(mk_arr, device_uid, device)
  File "/home/desktop/Downloads/razerCommander/macro_logic.py", line 14, in __init__
    self.set_all_current_macros()
  File "/home/desktop/Downloads/razerCommander/macro_logic.py", line 32, in set_all_current_macros
    key.set_current_macro(self.device)
  File "/home/desktop/Downloads/razerCommander/macro_logic.py", line 50, in set_current_macro
    self._set_macro(device, self.macro)
  File "/home/desktop/Downloads/razerCommander/macro_logic.py", line 43, in _set_macro
    device.macro.del_macro(self.key)
AttributeError: 'RazerDevice' object has no attribute 'macro'
```

Maybe the OrbWeaver doesn't support macros? I can't remember and have to start going more and more afk for today, but I think it might not.

My original issue was with keybindings, the ability to bind `01` to send `ESC` instead of the backtick. This is apparently on the driver to-do list so it might not happen any time soon, nor would I expect the macro system to do the same thing, I'm afraid... :(

GabMus, 2017-02-11:
Alright, this looks like a driver problem. Had this problem with the device simulator, I thought it was a problem with the simulator alone, but at this point I think it's more with the driver itself. I'll get @terrycain to have a look at this. And thanks for your help, it's been really important to me :)

mjtorn, 2017-02-11:
My pleasure and thank you for the work; sorry I don't have more time to delve into the actual code or anything.

ecazorg, 2017-06-26:
https://pastebin.com/kzvQg4kn

Similar issue. We think this is a razer-daemon/driver issue?

GabMus, 2017-10-26:
Does the Orbweaver work now?

GabMus, 2017-12-10:
the latest driver update should have fixed macro related stuff. please update everything and test it out again.

Altoholic, 2018-10-23:
> Tartarus-like devices support is kinda weird right now in razercommander, the tartarus is somewhat supported, the orbweaver should work as well

By "somewhat supported", you mean the lights only, not the remapping ?

As the tartarus bug is still open in openrazer github, i suppose i should not even try to use razercommander for now ?

alembiq, 2022-10-17:
Hi, 
is this still an issue in openrazer? i can't find it there ;( 

I'm jus trying to use my orbweaver chroma and the only thing i can do is to set up lights, not keys :(