GabMus/razerCommander #22: broken Ripple on Late 2016 RBS Ubuntu 10.16

nickdesaulniers, 2017-01-18:
Installed from the .deb release:

```
Traceback (most recent call last):
  File "/usr/share/razercommander/main.py", line 607, in on_fxListBox_row_selected
    if row.value in settingsPanes.keys():
AttributeError: 'NoneType' object has no attribute 'value'
```

GabMus, 2017-01-18:
I haven't been updating the deb in a while, try cloning the repo and
running the main.py file manually

nickdesaulniers, 2017-01-19:
```sh
nick@nick-Blade-Stealth:~$ git clone https://github.com/GabMus/razerCommander.git
nick@nick-Blade-Stealth:~$ cd razerCommander/
nick@nick-Blade-Stealth:~/razerCommander$ python main.py 
Traceback (most recent call last):
  File "main.py", line 2, in <module>
    import gi
ImportError: No module named gi
nick@nick-Blade-Stealth:~/razerCommander$ pip install gi
The program 'pip' is currently not installed. You can install it by typing:
sudo apt install python-pip
nick@nick-Blade-Stealth:~/razerCommander$ sudo apt install python-pip
nick@nick-Blade-Stealth:~/razerCommander$ pip install gi
You are using pip version 8.1.2, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
nick@nick-Blade-Stealth:~/razerCommander$ pip install --upgrade pip
nick@nick-Blade-Stealth:~/razerCommander$ python main.py 
Traceback (most recent call last):
  File "main.py", line 3, in <module>
    gi.require_version("Gtk", "3.0")
AttributeError: 'module' object has no attribute 'require_version'
```

???

nickdesaulniers, 2017-01-19:
looks like I need to be running python3.  Can you please update the readme to sugguest that Ubuntu users *not* download the old release?  Instead, I needed to:

```
$ sudo apt-get install python3 python-pip git python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
$ sudo pip install device gi
$ git clone https://github.com/GabMus/razerCommander.git
$ cd razerCommander
$ python3 main.py
```

Ripple was still broken, but now the trace looks like:

```
Traceback (most recent call last):
  File "main.py", line 632, in on_fxListBox_row_selected
    if row.value in settingsPanes.keys():
AttributeError: 'NoneType' object has no attribute 'value'
```

similar but more up to date line number.

nickdesaulniers, 2017-01-22:
https://github.com/terrycain/razer-drivers/issues/166

GabMus, 2017-02-08:
Any news on this?

closed 2017-02-11 by GabMus