GabMus/razerCommander #23: no devices! razer blade stealth on ubuntu 16.10

supersambo, 2017-01-23:
Hi, 
I tried razerCommander on a Razer Blade Stealth on Ubuntu 16.10. Unfortunately it doesn't find any devices.
According to https://terrycain.github.io/razer-drivers/ the drivers are supposed to be working on the Razer Blade Stealth.

Any ideas?

GabMus, 2017-01-26:
open a terminal, run the following and give me all the output:

`python3`
`from razer import client as rclient`
`dm=rclient.DeviceManager()`
`for device in dm.devices: print(device.type)`
`<press enter again>`
`quit()`

supersambo, 2017-01-26:
This doesn't return anything :( 
```
>>> len(dm.devices)
0
```

I suppose this is related to the issue reported [here](https://github.com/terrycain/razer-drivers/issues/67). However the [solution proposed](https://github.com/terrycain/razer-drivers/issues/67#issuecomment-260419314) doesn't solve my problem :(

GabMus, 2017-01-26:
Look, this is no problem I can solve in razerCommander, it's a bug in the driver/daemon/lib stack. I suggest you to open a new issue here: https://github.com/terrycain/razer-drivers

closed 2017-01-26 by GabMus