GabMus/razerCommander #27: RazerMouse object has no attribute 'macro'

A-Metaphysical-Drama, 2017-02-23:
I get this error when starting razercommander
```
Traceback (most recent call last):
  File "/usr/bin/razercommander", line 74, in <module>
    refreshDevices()
  File "/usr/bin/razercommander", line 70, in refreshDevices
    initDevices()
  File "/usr/bin/razercommander", line 36, in initDevices
    newdev = device.Device(dev)
  File "/usr/share/razercommander/device.py", line 30, in __init__
    self.device)
  File "/usr/share/razercommander/macro_logic.py", line 73, in make_device
    n_device = MacroDevice(mk_arr, device_uid, device)
  File "/usr/share/razercommander/macro_logic.py", line 14, in __init__
    self.set_all_current_macros()
  File "/usr/share/razercommander/macro_logic.py", line 32, in set_all_current_macros
    key.set_current_macro(self.device)
  File "/usr/share/razercommander/macro_logic.py", line 50, in set_current_macro
    self._set_macro(device, self.macro)
  File "/usr/share/razercommander/macro_logic.py", line 43, in _set_macro
    device.macro.del_macro(self.key)
AttributeError: 'RazerMouse' object has no attribute 'macro'
```

I'm on Archlinux using razercommander-git-1.1.0.r3.g17407d6-1
Razer Naga Hex v2

GabMus, 2017-02-23:
@terrycain should see this

terrycain, 2017-02-24:
Am aware of this

SoGCuicui, 2017-03-17:
Hi,
Exactly the same problem here, on Ubuntu 16.04 with a Naga Hex (lsusb gives me the reference "1532:0036") and a Naga Hex V2 ("1532:0050").
I am so sick of using a "xdotool" pseudo-solution which does not meet all my requirements...

terrycain, 2017-03-17:
Think ive fixed this, just not released yet

CavemanNinja, 2017-04-10:
+1 Ubuntu 16.10, razer naga hex v2

GabMus, 2017-04-13:
just tested the simulator with the latest git version of the driver and the naga hex v2 seems to work fine

SoGCuicui, 2017-04-13:
Great news! :)
I cannot test it right now, I will this evening.

GabMus, 2017-04-13:
@SoGCuicui alright, let me know

SoGCuicui, 2017-04-13:
Got:
```
Traceback (most recent call last):
  File "/usr/bin/razer-service", line 9, in <module>
    from razer_daemon.daemon import daemonize
ImportError: No module named 'razer_daemon'
```
after installed the driver from sources, which obviously wont starts.

I don't have enough energy to search what's going on for tonight... ^^'
I just putted back the ppa version.

GabMus, 2017-04-14:
@SoGCuicui not sure what happened here... Anyway, you should not include brackets, you should write d.fx. and then press the tab button twice to show suggestions

SoGCuicui, 2017-04-25:
Hi,
A few days has passed since I noticed that the Razer daemon has been upgraded with apt.
I tested to launch razerCommander right now and it recognizing my Naga v2! I can change the color of the macro buttons, as well as their functions!

I am just wondering if we will be able to asign "previous" and "next" function to the wheel? As you may noticed, in this model the wheel can be moved left and right, but it does nothing as default on my Ubuntu, sadly.

Thank you for this awesome work!

terrycain, 2017-04-26:
Soon. The left and right is a complicated issue

GabMus, 2017-10-26:
any news on this?

kgara, 2018-12-16:
Naga Hex v.2
BlackWidow Ultimate 2013
razercommander_1.2.1.1ubuntu1_all.deb
Also missing Macro section for both mouse and the keyboard. It was present in version V1.1.0.
Will appreciate help.

kgara, 2018-12-25:
Guys, any suspects or chances that it might be fixed? Thanks in advance.