GabMus/razerCommander #28: Razer DeathAdder Elite Support?

lswest, 2017-03-05:
I have a Razer DeathAdder Elite, which is recognized (at least, it appears in the device selector correctly). However, no lighting effects are listed besides Custom, and Brightness does nothing.

I have no idea what information might be necessary/useful to get the mouse lighting effects working properly, but I'm happy to supply any information that may be needed.

Specifically, the errors are:
```
WARNING:root:Brightness is not available
(0, 0, 0)
(0, 0, 0)
(0, 0, 0)
Traceback (most recent call last):
  File "/usr/bin/razercommander", line 461, in on_universalApplyButton_clicked
    enableFXwSettings(currentFX)
  File "/usr/bin/razercommander", line 351, in enableFXwSettings
    myrazerkb.applyCustom(rkb)
  File "/usr/share/razercommander/device.py", line 230, in applyCustom
    rindex, cindex, self._make_color_tuple(key.color)
  File "/usr/lib/python3.6/site-packages/razer/client/fx.py", line 870, in set
    self.__setitem__((y, x), rgb)
  File "/usr/lib/python3.6/site-packages/razer/client/fx.py", line 831, in __setitem__
    assert 0 <= key[1] < self._cols, "Column out of bounds"
AssertionError: Column out of bounds
```

GabMus, 2017-03-22:
Is this mouse supported by the driver stack?

lswest, 2017-03-22:
It works with Polychromatic Controller, which as far as I know uses the same driver?

GabMus, 2017-03-22:
@lswest yes it does. well, gotta check my frontend then :)

lswest, 2017-03-22:
@GabMus let me know if there's anything I should test out or any information you need me to pass on.

GabMus, 2017-03-23:
Look, I'm not quite sure where the problem is here. You should do me a favor: you should make a virtual machine with any distro of your choice (preferably arch or ubuntu based), passthrough your mouse to the vm and open an ssh session I can access to in said vm. This way I can test the mouse behavior and see what's wrong. Unfortunately I don't have any other way of testing this, since I don't have a razer mouse of any kind and the device simulator isn't working fine as of now. If you like you can contact me via email or through Telegram (username: @gabmus). Thank you very much for your support!

GabMus, 2017-04-13:
Hey there sorry if I'm replying now, but I've been busy lately. So, if you could run these commands in the terminal and send the output it'd be of great help (make sure your only connected razer device is the deathadder elite):
```
python3
from razer import client as rclient
dm=rclient.DeviceManager()
d=dm.devices[0]
print(d.capabilities)
quit()
```

lswest, 2017-04-13:
The output is as follows:

```
{'name': True, 'type': True, 'firmware_version': True, 'serial': True, 'brightness': False, 'macro_logic': False, 'lighting': True, 'lighting_breath_single': False, 'lighting_breath_dual': False, 'lighting_breath_triple': False, 'lighting_breath_random': False, 'lighting_wave': False, 'lighting_reactive': False, 'lighting_none': False, 'lighting_spectrum': False, 'lighting_static': False, 'lighting_ripple': False, 'lighting_ripple_random': False, 'lighting_pulsate': False, 'lighting_led_matrix': True, 'lighting_led_single': False, 'lighting_logo': True, 'lighting_logo_active': False, 'lighting_logo_blinking': False, 'lighting_logo_brightness': True, 'lighting_logo_pulsate': False, 'lighting_logo_spectrum': True, 'lighting_logo_static': True, 'lighting_logo_none': True, 'lighting_logo_reactive': True, 'lighting_logo_breath_single': True, 'lighting_logo_breath_dual': True, 'lighting_logo_breath_random': True, 'lighting_scroll': True, 'lighting_scroll_active': False, 'lighting_scroll_blinking': False, 'lighting_scroll_brightness': True, 'lighting_scroll_pulsate': False, 'lighting_scroll_spectrum': True, 'lighting_scroll_static': True, 'lighting_scroll_none': True, 'lighting_scroll_reactive': True, 'lighting_scroll_breath_single': True, 'lighting_scroll_breath_dual': True, 'lighting_scroll_breath_random': True, 'poll_rate': True, 'dpi': True}
```

A more legible version is [here](https://pastebin.com/wK7rEXqv)

GabMus, 2017-04-13:
@lswest thank you! so apparently mice have different effect names than keyboards... Now it's only a matter of implementing the right functions for the "new" effects. I'll try to reuse the code I used for keyboards, hopefully it will work. I'll keep you updated!

GabMus, 2017-04-13:
Try sending the output of:

```
python3
from razer import client as rclient
dm=rclient.DeviceManager()
d=dm.devices[0]
d.fx.(press tab twice)
```

lswest, 2017-04-13:
Am I supposed to actually replace the (press tab twice) section with two tabs? If so, I get this error:

```
  File "test2.py", line 4
    d.fx.(      )
         ^
SyntaxError: invalid syntax
```

I would assume it's supposed to be a string, which requires quotes. I've also tried running it as was listed above, which results in a SyntaxError as well. I've also tried assuming fx was a function and removed the period afterwards, but that returns `TypeError: 'RazerFX' object is not callable`

GabMus, 2017-04-13:
@lswest You actually have to press the tab key in your keyboard twice haha sorry if it wasn't clear

lswest, 2017-04-13:
@GabMus I assumed as much - are the brackets supposed to remain or not though?

GabMus, 2017-09-17:
Please, test this issue again running the latest version I just pushed on `master`. It's already available in the AUR, with other distro packages coming soon.

lswest, 2017-09-17:
I just re-installed the razercommand-git package from AUR, and now receive the following error when trying to launch it:

`Exception: No module named 'openrazer'`

Following razer-related packages are installed:
```
aur/razercfg 0.39-1 (43 | 0.22) [installed]
aur/python-razer 1.1.16-1 (13 | 1.63) [installed]
aur/razer-daemon 1.1.16-1 (13 | 1.63) [installed]
aur/razer-driver-dkms 1.1.16-1 (13 | 1.63) [installed]
aur/openrazer-meta 1.1.16-1 (13 | 1.63) [installed]
aur/polychromatic 0.3.10-1 (4 | 0.78) [installed]
aur/razercommander-git 1.2.0.r31.g760e251-1 (3 | 0.06) [installed]
```

The entire error message is as follows:
```

** (razercommander:4749): WARNING **: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-7m66E546Zm: Connection refused
Importing device logic, waiting for daemon
ERROR: the daemon is not responding!
Try running `killall razer-daemon && razer-daemon` or rebooting. If this doesn't work, please fill an issue!
Exception: No module named 'openrazer'
```

I've already rebooted and killed/started the razer-daemon. Polychromatic controller still works as expected.

GabMus, 2017-09-17:
Try installing razer-driver-dkms-git razer-daemon-git and python-razer-git
from AUR

lswest, 2017-09-17:
Thanks! After switching to the -git versions of everything, razercommander works. I've got access to the various lighting effects, and can set brightness, colors, DPI, etc.

GabMus, 2017-09-17:
Cool, have fun!

closed 2017-09-17 by GabMus