GabMus/razerCommander #31: Unresolved Dependencies for Arch Linux

rudolfschmidt, 2017-05-05:
Hi,

Installing razercommander-git  from aur via yaourt

`$ yaourt -S razercommander-git`

I get

```
==> Continue installing razer-drivers ? [Y/n]
==> [v]iew package contents [c]heck package with namcap
==> ---------------------------------------------------
==> y

loading packages...
resolving dependencies...
warning: cannot resolve "python-notify2", a dependency of "razer-daemon"
warning: cannot resolve "razer-daemon", a dependency of "openrazer-meta"
warning: cannot resolve "python-notify2", a dependency of "razer-daemon"
warning: cannot resolve "razer-daemon", a dependency of "python-razer"
warning: cannot resolve "python-razer", a dependency of "openrazer-meta"
warning: cannot resolve "python-notify2", a dependency of "razer-daemon"
warning: cannot resolve "razer-daemon", a dependency of "python-razer"
warning: cannot resolve "python-notify2", a dependency of "razer-daemon"
:: The following packages cannot be upgraded due to unresolvable dependencies:
      openrazer-meta  python-razer  razer-daemon

:: Do you want to skip the above packages for this upgrade? [y/N] 
```

Means there is no way to get the required dependencies.

Get you tell where we get those dependencies for arch linux? the official razer drivers are already installed by manual installation from the source code (no pacman package). Maybe thats the problem?

Thank you

hobarrera, 2017-05-05:
This is vey clearly related to the packaging of `razer-daemon` and other packages on ArchLinux.

The tools you're using are unrelated to this project, and the package is maintained elsewhere. You should bring that up in the AUR package's comment page.

GabMus, 2017-05-05:
> the official razer drivers are already installed by manual installation from the source code

Yeah, that's the problem. So either uninstall the driver and use the packages instead, or edit the razercommander pkgbuild to remove the driver dependencies, or clone the git version of razercommander and run it manually

rudolfschmidt, 2017-05-05:
Thanks for answer,
I did exactly that, I have cloned the repo and run the main.py. 
I get the message that no device is found. Can you help me to enable the device for your configuration tool?

GabMus, 2017-05-06:
@rudolfschmidt What device do you have?

rudolfschmidt, 2017-05-06:
I have the Razer DeathAddr Chroma but I have found the error. The problem was a wrong installation of the razer drivers. I have used the source codes but the provider of the source codes recommended me to use the aur package that have worked. Now your app find my device successfully.  

The problem what I have now is that your app just show the following screen: https://clbin.com/7071Eg.png

Are not other options available?

GabMus, 2017-09-17:
This is now fixed in the latest version I just pushed on `master`. It's already available in the AUR, with other distro packages coming soon

closed 2017-09-17 by GabMus