GabMus/razerCommander #35: ValueError: Key 10 is not in razer.keyboard.KEY_MAPPING

vozeldr, 2017-07-30:
I have 5 razer devices, 4 of which are supported by razer-drivers, which I have installed as described here: https://terrycain.github.io/razer-drivers/#ubuntu

lsusb | grep '1532:'
Bus 005 Device 003: ID 1532:0208 Razer USA, Ltd 
Bus 005 Device 002: ID 1532:0504 Razer USA, Ltd 
Bus 003 Device 003: ID 1532:003e Razer USA, Ltd 
Bus 003 Device 002: ID 1532:0203 Razer USA, Ltd 
Bus 003 Device 007: ID 1532:0c00 Razer USA, Ltd

After rebooting from the razer-drivers installation, I cloned razerCommander and upon running main.py get the following error:

./main.py 
Importing device logic, waiting for daemon
Device logic loaded, deamon is alive
Traceback (most recent call last):
  File "./main.py", line 82, in <module>
    refreshDevices()
  File "./main.py", line 78, in refreshDevices
    initDevices()
  File "./main.py", line 44, in initDevices
    newdev = device.Device(dev)
  File "/home/asdf/Projects/razerCommander/device.py", line 30, in __init__
    self.device)
  File "/home/asdf/Projects/razerCommander/macro_logic.py", line 73, in make_device
    n_device = MacroDevice(mk_arr, device_uid, device)
  File "/home/asdf/Projects/razerCommander/macro_logic.py", line 14, in __init__
    self.set_all_current_macros()
  File "/home/asdf/Projects/razerCommander/macro_logic.py", line 32, in set_all_current_macros
    key.set_current_macro(self.device)
  File "/home/asdf/Projects/razerCommander/macro_logic.py", line 50, in set_current_macro
    self._set_macro(device, self.macro)
  File "/home/asdf/Projects/razerCommander/macro_logic.py", line 43, in _set_macro
    device.macro.del_macro(self.key)
  File "/usr/lib/python3/dist-packages/razer/client/macro.py", line 66, in del_macro
    raise ValueError("Key {0} is not in razer.keyboard.KEY_MAPPING".format(bind_key))
ValueError: Key 10 is not in razer.keyboard.KEY_MAPPING

vozeldr, 2017-07-31:
Well, for some reason it's the Razer Tartarus Chroma causing the issue. I got your python to run by changing initDevices in main.py to:
```
def initDevices():
    global devicesList
    devicesList = []
    for dev in device.devlist:
        try:
            newdev = device.Device(dev)
            devicesList.append(newdev)
        except:
            print('Skipping device: ' + dev.name)
            pass
```
After running that, I see "Skipping device: Razer Tartarus Chroma" in the console when I run it and the application works OK for changing the colors to custom profiles (although the UI is a bit difficult) on my Razer Blackwidow Chroma. I haven't tried to do anything with the Firefly or Kraken headset.

Also, a misleading bit of code you have in main.py is a message refering to razer-service but I believe it should be razer-daemon.

GabMus, 2017-10-26:
Unfortunately it's not easy for me to support devices I don't have. I can do my best to implement support for them using the device simulators but those aren't perfect. Please tell me if you still have problems with your Tartarus and maybe test the firefly and the Kraken.

Also, why do you think the profiles UI is difficult, and how do you think I could improve it?

closed 2017-11-29 by GabMus

vozeldr, 2017-11-29:
Oh, yeah... after you merged (and made some tweaks) to the change I submitted, I no longer get a crash. It still doesn't recognize the Tartarus but I guess that's just a problem with the open drivers and not your commander.