GabMus/razerCommander #37: razerCommander not detecting any devices

JokerakaXRAY, 2017-08-08:
First of all I am on Antergos GNOME with a Deathadder Chroma, Blackwidow Chroma and Kraken Chroma.

I already tried multiple AUR helpers to install the razer drivers but none of them seem to work.

I tried pacaur and yaourt with yaourt having installed my current packages.

I will apologize in advance for the kind of bad formating.

Starting main.py results in:
```
[joker@Joker-GNOME razercommander]$ ./main.py 
Importing device logic, waiting for daemon
Device logic loaded, deamon is alive
no devices
no devices
```

Running "lsusb | grep '1532:'" to detect devices results in:
```
[joker@Joker-GNOME ~]$ lsusb | grep '1532:'
Bus 003 Device 004: ID 1532:0203 Razer USA, Ltd 
Bus 003 Device 006: ID 1532:0043 Razer USA, Ltd 
Bus 003 Device 005: ID 1532:0504 Razer USA, Ltd 
```

Razer Mount Log:
```
Aug 08 02:04:06 Joker-GNOME razer_mount[335]: Driver razerkraken
Aug 08 02:04:06 Joker-GNOME razer_mount[337]: Driver razermouse
Aug 08 02:04:06 Joker-GNOME razer_mount[339]: Driver razermouse
Aug 08 02:04:06 Joker-GNOME razer_mount[336]: Driver razermouse
Aug 08 02:04:06 Joker-GNOME systemd-udevd[259]: Process 'razer_mount razermouse 0003:1532:0043.0003' failed with exit code 1.
Aug 08 02:04:06 Joker-GNOME razer_mount[350]: Device_ID 0003:1532:0043.0001
Aug 08 02:04:06 Joker-GNOME systemd-udevd[252]: Process 'razer_mount razermouse 0003:1532:0043.0001' failed with exit code 1.
Aug 08 02:04:06 Joker-GNOME razer_mount[351]: Device_ID 0003:1532:0043.0003
Aug 08 02:04:06 Joker-GNOME systemd-udevd[241]: Process 'razer_mount razermouse 0003:1532:0043.0002' failed with exit code 1.
Aug 08 02:04:06 Joker-GNOME razer_mount[349]: Device_ID 0003:1532:0043.0002
Aug 08 02:04:06 Joker-GNOME systemd-udevd[254]: Process 'razer_mount razerkraken 0003:1532:0504.0007' failed with exit code 1.
Aug 08 02:04:06 Joker-GNOME razer_mount[352]: Device_ID 0003:1532:0504.0007
Aug 08 02:04:06 Joker-GNOME systemd-udevd[252]: Process 'razer_mount razerkbd 0003:1532:0203.0004' failed with exit code 1.
Aug 08 02:04:06 Joker-GNOME razer_mount[354]: Modprobing razermouse
Aug 08 02:04:06 Joker-GNOME systemd-udevd[241]: Process 'razer_mount razerkbd 0003:1532:0203.0005' failed with exit code 1.
Aug 08 02:04:06 Joker-GNOME razer_mount[353]: Modprobing razermouse
Aug 08 02:04:06 Joker-GNOME systemd-udevd[241]: Process 'razer_mount razerkbd 0003:1532:0203.0006' failed with exit code 1.
Aug 08 02:04:06 Joker-GNOME razer_mount[355]: Modprobing razermouse
Aug 08 02:04:06 Joker-GNOME razer_mount[356]: Modprobing razerkraken
Aug 08 02:04:06 Joker-GNOME razer_mount[362]: Driver razerkbd
Aug 08 02:04:06 Joker-GNOME razer_mount[363]: Device_ID 0003:1532:0203.0004
Aug 08 02:04:06 Joker-GNOME razer_mount[364]: Modprobing razerkbd
Aug 08 02:04:06 Joker-GNOME razer_mount[367]: Driver razerkbd
Aug 08 02:04:06 Joker-GNOME razer_mount[368]: Device_ID 0003:1532:0203.0005
Aug 08 02:04:06 Joker-GNOME razer_mount[369]: Modprobing razerkbd
Aug 08 02:04:06 Joker-GNOME razer_mount[372]: Driver razerkbd
Aug 08 02:04:06 Joker-GNOME razer_mount[373]: Device_ID 0003:1532:0203.0006
Aug 08 02:04:06 Joker-GNOME razer_mount[374]: Modprobing razerkbd
```

Opening the Openrazer log shows this:
```
2017-08-08 02:05:06 | razer                          | INFO     | Initialising Daemon (v1.1.15). Pid: 1575
2017-08-08 02:05:06 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2012 --------------------- (1532:010D)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
[... 26 lines cut ...]
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerNaga2014 --------------------------- (1532:0040)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerNagaHex ---------------------------- (1532:0041)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexRed ------------------------- (1532:0036)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexV2 -------------------------- (1532:0050)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerNostromo --------------------------- (1532:0111)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaver -------------------------- (1532:0113)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaverChroma -------------------- (1532:0207)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerOrnata ----------------------------- (1532:021F)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerOrochi2011 ------------------------- (1532:0013)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerOrochi2013 ------------------------- (1532:0039)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerOuroboros -------------------------- (1532:0032)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerTaipan ----------------------------- (1532:0034)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0201)
2017-08-08 02:05:06 | razer                          | DEBUG    | Loaded device specification: RazerTartarusChroma --------------------- (1532:0208)
2017-08-08 02:05:06 | razer                          | INFO     | Adding razer.devices.getDevices method to DBus
2017-08-08 02:05:06 | razer                          | INFO     | Adding razer.daemon.supportedDevices method to DBus
2017-08-08 02:05:06 | razer                          | INFO     | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2017-08-08 02:05:06 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2017-08-08 02:05:06 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2017-08-08 02:05:06 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2017-08-08 02:05:06 | razer                          | INFO     | Adding razer.daemon.version method to DBus
```

GabMus, 2017-09-17:
Please, test this issue again running the latest version I just pushed on `master`. It's already available in the AUR, with other distro packages coming soon.

DarthHail, 2017-10-24:
Having the same issue, razer is showing up in devices list and in journalctl when plugged in, but still getting a no devices found error:

` /usr/lib/gdm3/gdm-x-session[843]: (II) Using input driver 'libinput' for 'Razer Razer BlackWidow Chroma'
Oct 24 10:44:20  /usr/lib/gdm3/gdm-x-session[843]: (II) systemd-logind: got fd for /dev/input/event4 13:68 fd 53 paused 0
Oct 24 10:44:20  /usr/lib/gdm3/gdm-x-session[843]: (**) Razer Razer BlackWidow Chroma: always reports core events
Oct 24 10:44:20  /usr/lib/gdm3/gdm-x-session[843]: (**) Option "Device" "/dev/input/event4"
Oct 24 10:44:20  /usr/lib/gdm3/gdm-x-session[843]: (**) Option "_source" "server/udev"
Oct 24 10:44:20  /usr/lib/gdm3/gdm-x-session[843]: (II) input device 'Razer Razer BlackWidow Chroma', /dev/input/event4 is tagged by udev as: Mouse
Oct 24 10:44:20  /usr/lib/gdm3/gdm-x-session[843]: (II) input device 'Razer Razer BlackWidow Chroma', /dev/input/event4 is a pointer caps
Oct 24 10:44:20  /usr/lib/gdm3/gdm-x-session[843]: (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.5/2-1.5:1.2/0003:1532:0203.0009/input/input19/event4"
Oct 24 10:44:20  /usr/lib/gdm3/gdm-x-session[843]: (II) XINPUT: Adding extended input device "Razer Razer BlackWidow Chroma" (type: MOUSE, id 14)
Oct 24 10:44:20  /usr/lib/gdm3/gdm-x-session[843]: (**) Option "AccelerationScheme" "none"
Oct 24 10:44:20  /usr/lib/gdm3/gdm-x-session[843]: (**) Razer Razer BlackWidow Chroma: (accel) selected scheme none/0
Oct 24 10:44:20  /usr/lib/gdm3/gdm-x-session[843]: (**) Razer Razer BlackWidow Chroma: (accel) acceleration factor: 2.000
Oct 24 10:44:20  /usr/lib/gdm3/gdm-x-session[843]: (**) Razer Razer BlackWidow Chroma: (accel) acceleration threshold: 4
Oct 24 10:44:20  /usr/lib/gdm3/gdm-x-session[843]: (II) input device 'Razer Razer BlackWidow Chroma', /dev/input/event4 is tagged by udev as: Mouse
Oct 24 10:44:20  /usr/lib/gdm3/gdm-x-session[843]: (II) input device 'Razer Razer BlackWidow Chroma', /dev/input/event4 is a pointer caps
Oct 24 10:44:32  org.gabmus.razercommander.desktop[23637]: Importing device logic, waiting for daemon
Oct 24 10:44:32  org.gabmus.razercommander.desktop[23637]: Device logic loaded, daemon is alive
Oct 24 10:44:32  org.gabmus.razercommander.desktop[23637]: no devices
`

Module shows, but does not detect any device. Currently on most recent version 1.2.0

GabMus, 2017-10-25:
Pushed a change that maybe should make the supported devices work. Please test the latest version from master and report back. Unfortunately I don't have a Kraken headset, so I can't test it out, along with many other devices I don't have. Of course, if anyone wants to donate one, I'd be glad to add support for them 😃

GabMus, 2017-12-10:
closing for inactivity. as always, if you have any news on this, feel free to reopen

closed 2017-12-10 by GabMus

gun1x, 2018-08-18:
I am experiencing the same issue on Fedora 28, but in my case, it never gets to modprobe the razerkbd:

```
 INSERT  gunix  ~  docs  sudo journalctl -b | grep -i razer_mount
Aug 18 17:50:29 localhost.localdomain razer_mount[2191]: Driver razerkbd
Aug 18 17:50:29 localhost.localdomain razer_mount[2193]: Driver razerkbd
Aug 18 17:50:29 localhost.localdomain razer_mount[2195]: Device_ID 0003:1532:0209.0003
Aug 18 17:50:29 localhost.localdomain razer_mount[2194]: Device_ID 0003:1532:0209.0002
Aug 18 17:50:29 localhost.localdomain razer_mount[2196]: Driver razerkbd
Aug 18 17:50:29 localhost.localdomain razer_mount[2200]: Modprobing razerkbd
Aug 18 17:50:29 localhost.localdomain razer_mount[2202]: Device_ID 0003:1532:0209.0001
Aug 18 17:50:29 localhost.localdomain razer_mount[2197]: Modprobing razerkbd
Aug 18 17:50:29 localhost.localdomain systemd-udevd[2182]: Process 'razer_mount razerkbd 0003:1532:0209.0002' failed with exit code 1.
Aug 18 17:50:29 localhost.localdomain razer_mount[2206]: Modprobing razerkbd
Aug 18 17:50:29 localhost.localdomain systemd-udevd[2172]: Process 'razer_mount razerkbd 0003:1532:0209.0003' failed with exit code 1.
Aug 18 17:50:29 localhost.localdomain systemd-udevd[2171]: Process 'razer_mount razerkbd 0003:1532:0209.0001' failed with exit code 1.
```

commit 8160debe (2017-10-25): Trying to fix #37