GabMus/razerCommander #38: Error when trying to use custom mode on Razer Blade 2016

lpaine, 2017-08-25:
Traceback (most recent call last):
  File "/usr/bin/razercommander", line 469, in on_universalApplyButton_clicked
    enableFXwSettings(currentFX)
  File "/usr/bin/razercommander", line 359, in enableFXwSettings
    myrazerkb.applyCustom(rkb)
  File "/usr/share/razercommander/device.py", line 230, in applyCustom
    rindex, cindex, self._make_color_tuple(key.color)
  File "/usr/lib/python3.6/site-packages/razer/client/fx.py", line 997, in set
    self.__setitem__((y, x), rgb)
  File "/usr/lib/python3.6/site-packages/razer/client/fx.py", line 958, in __setitem__
    assert 0 <= key[1] < self._cols, "Column out of bounds"
AssertionError: Column out of bounds

GabMus, 2017-09-17:
Please, test this issue again running the latest version I just pushed on `master`. It's already available in the AUR, with other distro packages coming soon.

GabMus, 2017-12-10:
closing for inactivity

closed 2017-12-10 by GabMus