GabMus/razerCommander #39: dbus Error

m1roh, 2017-08-29:
Hi, i tried to install razerCommander for maybe 10 times but everytime when I try to start it in the terminal with razercommander I get this : 
`ERROR:dbus.proxies:Introspect error on :1.3:/org/razer: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying
DEBUG:dbus.proxies:Executing introspect queue due to error
Traceback (most recent call last):
  File "/usr/share/razercommander/main.py", line 7, in <module>
    import device
  File "/usr/share/razercommander/device.py", line 11, in <module>
    device_manager = rclient.DeviceManager()
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 31, in __init__
    self._device_serials = self._dbus_devices.getDevices()
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name :1.3 was not provided by any .service files
`
is there any solution to solve this?

hobarrera, 2017-08-29:
Are the [razer drivers](https://github.com/terrycain/razer-drivers/) installed?

m1roh, 2017-08-29:
Hi hobarrera, of course they're installed! What is really weird it's it use to worked when I my laptop was in dual-boot with win 10 but I made a clean install with only ubuntu today and it's doesn't work anymore...

m1roh, 2017-09-02:
I finally solved the problem. In fact the last time I turned off my computer before deleting Windows, I was logged on my razer account with my "non-lighting profile". So it's look like something, somewhere, remembered that. So I just pluggged my keyboard on a windows computer, installed the official driver, logged on my razer synapse account and move on my "lighting profile", then I reppluged this on my ubuntu laptop and... magic happens ! It works !!!!! Thank you for your answers...

closed 2017-09-02 by m1roh