GabMus/razerCommander #40: Razer Tartarus detected as "Unknown Device" and Keys cannot be rebound.

DreadPirateLynx, 2017-09-01:


GabMus, 2017-09-02:
Try installing the new version from the rebase branch

DreadPirateLynx, 2017-09-04:
Tried to build it, but got the following error:
`Meson encountered an error in file data/meson.build, line 75, column 0:
Native dependency 'glib-2.0' not found
`
As far as I can tell, glib-2.0 is installed on my system.

GabMus, 2017-09-05:
If you're using Ubuntu there's probably a -dev version of that library, you
most likely need that.

DreadPirateLynx, 2017-09-05:
Hmmm...

`sudo apt-get install glib-2.0-dev -y
[sudo] password for dreadpiratelynx: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libspice-client-glib-2.0-dev' for regex 'glib-2.0-dev'
libspice-client-glib-2.0-dev is already the newest version (0.30-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.`

GabMus, 2017-09-17:
Please, test this issue again running the latest version I just pushed on `master`. It's already available in the AUR, with other distro packages coming soon.

GabMus, 2017-10-26:
Any news on this?

GabMus, 2017-11-28:
Closing for inactivity

panzth, 2017-12-14:
New here. As of now, one of the last updates of the daemon made Tartarus Chroma unrecognizable by the daemon, so can't test for the time being. Had some problems earlier compiling in Fedora but latest source compiled very easily. Will inform you when tartarus gets along with the daemon once more.

GabMus, 2017-12-15:
Guess I have to reopen this then

panzth, 2017-12-15:
Managed to get everything working. Polychromatic recognizes Tartarus and can change  colors, brightness, etc.. Razer Commander doesn't.
In console i get this message,

Skipping device Razer Tartarus Chroma
  Exception: Key 10 is not in openrazer.keyboard.KEY_MAPPING

GabMus, 2017-12-15:
razerCommander manages macros, while polychromatic (as far as I remeber) doesn't. razerCommander skips the tartarus because macro stuff is currently broken in the driver stack.
talking of wich, @terrycain have a look.

terrycain, 2017-12-15:
Yeah in the daily packages we're disabling macro for now as its causing a
fair amount of issues

GabMus, 2017-12-16:
@panzth you should now be able to use basic functionalities of razerCommander even if the macro logic in the driver stack is currently broken, and you will see a slightly different message. Hope this helps a little bit

panzth, 2017-12-16:
Now it functions properly. Still no macro option, but as i understand, we have to wait for macros to be implemented in openrazer first.

panzth, 2017-12-27:
With the latest 2.2 version of openrazer macros work!! Although not really usable but one step closer.

When pressing a button it records 2 keystrokes! That's the deal-breaker so far.
Also, it seems that modifier keys cannot be set as buttons(as a single button, eg try the super key)
Other than that, the only downside is that for now there is no  support for more than 1 sets of keys and/or the ability to change between different macro sets.

GabMus, 2018-01-06:
> When pressing a button it records 2 keystrokes

What do you mean? Could you elaborate?

> Also, it seems that modifier keys cannot be set as buttons

For this you can set a command macro with the following code: `xdotool key Super`

panzth, 2018-01-06:
When a button is pressed keystroke is recorded and when the button is released another keystroke is recorded. So, every time a key is pressed it records at least 2 strokes. Did i explain it better?

Also i don't know if holding the button down works. For example i set the command "xdotool key Shift" but holding the button and pressing a letter didn't result in capital letter.

If it's not possible to have multiple keysets, maybe saving and loading presets is possible? 
If not, can razercommander keep settings between reboots?

Hope it's not too difficult to fix those issues. Anyway thanks for your work and time and if there is anything else i can do to help, let me know.

GabMus, 2018-01-06:
> Did i explain it better?

Yes, definitely. Unfortunately I never tested this feature extensively since I don't own any macro enabled razer devices.

> i don't know if holding the button down works

I don't think it does, `xdotool key` just rapidly presses and releases they key. There should be an option to simulate holding, and maybe if you explore the xdotool man page you could find an option to set the macro key as a toggle. I have to do some work on macros apparently, but I'm not sure if I'll be able to do anything from my end.

> If it's not possible to have multiple keysets, maybe saving and loading presets is possible?

It's very likely, I could do something similar to what I've done with custom color profiles. That'd require some work tho.

> can razercommander keep settings between reboots?

Kinda, you should launch razercommander as a startup application. You can make a custom launcher that runs `razercommander -q` so that razercommander starts, inits macros and quits, and set that as a startup app instead.

> Anyway thanks for your work and time and if there is anything else i can do to help, let me know.

You're welcome! I really enjoy developing razercommander, but it's a really big project and requires a lot of effort. If you want to help me out, you can donate a device you want better support for, that's already a huge help!
Otherwise, you could donate me a coffee or something, if you feel like it. You can use my email and send me a couple of bucks via paypal or if you prefer I can send you the address of a cryptocurrency wallet of your choice. :grin:

GabMus, 2018-01-10:
@panzth I think I need a better explanation on this issue:

> When a button is pressed keystroke is recorded and when the button is released another keystroke is recorded. So, every time a key is pressed it records at least 2 strokes.

I just tested macro recording and it seems to be working fine. So I think I misunderstood your problem. Tell me if the following is correct: *I assign the key 5 on the tartarus to simulate the keystroke `a+s+d`. I open up a text editor, so I press and hold the 5 key and I see `asd` on the text editor. I release the 5 key and I see another `asd` written on the text editor*.

Hope I got that right this time. If not, please record a video showing the issue and send it here.

panzth, 2018-01-10:
Yeah, that's right!

DreadPirateLynx, 2018-03-22:
Sorry I disappeared when you originally put forward a fix; I had decided to use device specific keyboard layouts using xkb to get my Tartarus working since I wasn't ever able to get the fix installed. So on to the current problem:

Unfortunately, I no longer have a Linux computer, and due to some issues with my Windows license, I can't risk damaging my installation with a duel boot because I'll be unable to reinstall it. This of course means that I won't be able to do any testing, but when I looked over the messages for this issue today, I noticed something that may be of help:

`xdotool key` simulates a key press and release. `xdotool keydown` simulates holding a key down, and `xdotool keyup` simulates the release. It sounds like the software isn't distinguishing between the user pressing and releasing the key, instead counting each as both. Then your `xdotool key` command is telling it to press and release the key twice because of it.