GabMus/razerCommander #42: Configuration applied to all devices (mouse, keyboard and headset)

Ayak973, 2017-09-13:
Hello,
I'm using Debian 9.1, razer-daemon 1.1.16-0ubuntu1 and have installed razerCommander application from your repository (the .deb version 1.1.0).
I can select my 3 devices from razerCommander (Razer Mamba Tournament Edition, Razer Kraken 7.1v2 and Razer BlackWidow chroma v2), but when I apply the settings (color, light effect...) for one device, it set the same settings for all the others. 
Also when i select my mouse, it shows the keyboard layout, as well as when i select my headset.
Is this because of an unsupported hardware version?
How can I help you to diagnose or repair this issue?

GabMus, 2017-09-13:
@Ayak973 I fixed almost all of this in the new version in the rebase branch. You can try it out but you'll have to build it yourself

Ayak973, 2017-09-13:
Thank you, I'll give it a try as soon as possible. Would you want that I tell you if it works?
If I have understood well, I've to remove razer-daemon 1.1.16-0ubuntu1 and install openrazer-daemon?
Anyway, you've made a great application, many thanks !

GabMus, 2017-09-13:
Yep, and also install razercommander from the rebase branch

Ayak973, 2017-09-14:
I've successfully installed openrazer-daemon (I've used the .deb for ubuntu provided on Launchpad, with different workaroud to get it working - it's not recommanded, I know). The different modules are successfully loaded, and the openrazer daemon logs shows that it is running, hardware is properly detected, and it is in version 2.0.0.
I've cloned your repository (`git clone -b rebase https://github.com/GabMus/razerCommander.git` <- I'm not really sure for this step, I think I've downloaded the version you tell me), and build RazerCommander with meson & ninja (I get issues first because Debian Wheezy didnt have meson 0.4, only 0.37, and your project requires version 0.40, so I had to upgrade some packages).
When I launch openrazer, I've the following message :

> Importing device logic, waiting for daemon
> ERROR: the daemon is not responding!
> Try running `killall razer-daemon && razer-daemon` or rebooting. If this doesn't work, please fill an issue!
> Exception: No module named 'razer'

Is it because I've downloaded an incorrect version of RazerCommander?
Is it because RazerCommander is searching for razer-daemon instead of openrazer-daemon, or only because the error message has not been updated from the previous version (in that last case, it will tell me that the installation of the daemon is broken)?
Any hints will be appreciated. Please tell me if you need additional informations.

GabMus, 2017-09-14:
@Ayak973 > Is it because RazerCommander is searching for razer-daemon instead of openrazer-daemon

Most likely, yes. I have to update it

Ayak973, 2017-09-14:
Ok, if it's not too difficult I will try to read your code and modify it by hand. If not, I will wait. Can you please tell me when it could be released?

GabMus, 2017-09-14:
@Ayak973 I think you can edit it yourself, it's nothing difficult, just look in the main file. I'll update it whenever I can, I've been busy recently

Ayak973, 2017-09-14:
I've edited the file device.py, recompiled the project, and have another message from DBus (message timeout). I"ve tried to play with the openrazer driver scripts directly in `/sys/bus/hid/drivers/razerkbd/<ID>`, and cannot light on any devices. I think that the openrazer ubuntu driver 2.0.0 is not usable on my debian system. 
I've downgrade to razer-daemon, wait for the debian release of openrazer-daemon, and close the issue.
Thank you

closed 2017-09-14 by Ayak973