GabMus/razerCommander #43: Razer Naga cannot rebined side buttons

jambonmcyeah, 2017-09-14:


GabMus, 2017-09-17:
Please, test this issue again running the latest version I just pushed on `master`. It's already available in the AUR, with other distro packages coming soon.

jambonmcyeah, 2017-09-18:
Now it doesn't even start. :(
```
Importing device logic, waiting for daemon
ERROR: the daemon is not responding!
Try running `killall razer-daemon && razer-daemon` or rebooting. If this doesn't work, please fill an issue!
```

jambonmcyeah, 2017-09-18:
I `killall razer-daemon && razer-daemon` and rebooted

jambonmcyeah, 2017-09-18:
I updated again now the error message is this
```
Importing device logic, waiting for daemon
ERROR: the daemon is not responding!
Try running `killall razer-daemon && razer-daemon` or rebooting. If this doesn't work, please fill an issue!
Exception: No module named 'openrazer'
```

GabMus, 2017-09-19:
You probably need the very latest version of the driver. For Arch it's the
git version.

GabMus, 2017-11-28:
Any news on this?

jambonmcyeah, 2017-11-29:
The lights now work, but still can't rebind buttons

jambonmcyeah, 2017-11-29:
[image]
[image]

GabMus, 2017-11-29:
Just pushed a small update, you *should* be able to assign macro keys now. Waiting for an update on your end.

jambonmcyeah, 2017-11-29:
I installed the razercommander-git from pacaur and it won't even start now

GabMus, 2017-11-29:
weird. what version of the driver do you have installed?

GabMus, 2017-11-29:
by the way, can you give me the command line output of razercommander?

jambonmcyeah, 2017-11-29:
```
[jambon@jambonarch ~]$ razercommander 
Importing device logic, waiting for daemon
Device logic loaded, daemon is alive
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 245, in initDevices
    newdev = device.Device(dev)
  File "/usr/lib/python3.6/site-packages/razercommander/device.py", line 50, in __init__
    self.device)
  File "/usr/lib/python3.6/site-packages/razercommander/macro_logic.py", line 73, in make_device
    n_device = MacroDevice(mk_arr, device_uid, device)
  File "/usr/lib/python3.6/site-packages/razercommander/macro_logic.py", line 14, in __init__
    self.set_all_current_macros()
  File "/usr/lib/python3.6/site-packages/razercommander/macro_logic.py", line 32, in set_all_current_macros
    key.set_current_macro(self.device)
  File "/usr/lib/python3.6/site-packages/razercommander/macro_logic.py", line 50, in set_current_macro
    self._set_macro(device, self.macro)
  File "/usr/lib/python3.6/site-packages/razercommander/macro_logic.py", line 43, in _set_macro
    device.macro.del_macro(self.key)
  File "/usr/lib/python3.6/site-packages/openrazer/client/macro.py", line 66, in del_macro
    raise ValueError("Key {0} is not in openrazer.keyboard.KEY_MAPPING".format(bind_key))
ValueError: Key 10 is not in openrazer.keyboard.KEY_MAPPING

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 687, in do_command_line
    self.do_activate()
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 661, in do_activate
    self.refreshDevices()
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 302, in refreshDevices
    self.initDevices()
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 248, in initDevices
    print('Skipping device: {}'.format(device.name))
AttributeError: module 'razercommander.device' has no attribute 'name'
```

```
[jambon@jambonarch ~]$ pacman -Q | grep razer
openrazer-daemon 2.1.1-1
openrazer-driver-dkms 2.1.1-1
python-openrazer 2.1.1-1
razercommander-git 1.2.0.r11.g1ad7e58-1
```

GabMus, 2017-11-29:
Just pushed an experimental update, didn't update the AUR version myself, so just run `pacaur -S razercommander-git` if you use pacaur or the same thing with yaourt if you use that instead (or clone the AUR repo and `makepkg -s && sudo pacman -U *.tar.xz` manually if you prefer) and tell me if you can do anything at all.

The problem seems to be that the openrazer driver doesn't recognise the keys 10, 11 and 12. I tried masking those out, and it should work now I think. If it does, try assigning some keys and see if macros work.

jambonmcyeah, 2017-11-29:
It opens now but the macros still didn't work
```
[jambon@jambonarch razercommander-git]$ razercommander
Importing device logic, waiting for daemon
Device logic loaded, daemon is alive
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 687, in do_command_line
    self.do_activate()
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 670, in do_activate
    self.refreshFxList()
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 440, in refreshFxList
    ).set_upper(self.active_razer_device.get_max_dpi())
  File "/usr/lib/python3.6/site-packages/razercommander/device.py", line 477, in get_max_dpi
    return self.device.max_dpi
  File "/usr/lib/python3.6/site-packages/openrazer/client/devices/mice.py", line 30, in max_dpi
    return int(self._dbus_interfaces['dpi'].maxDPI())
  File "/usr/lib/python3.6/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3.6/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownMethod: Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/dbus/service.py", line 654, in _message_cb
    (candidate_method, parent_method) = _method_lookup(self, method_name, interface_name)
  File "/usr/lib/python3.6/site-packages/dbus/service.py", line 246, in _method_lookup
    raise UnknownMethodException('%s is not a valid method of interface %s' % (method_name, dbus_interface))
dbus.exceptions.UnknownMethodException: org.freedesktop.DBus.Error.UnknownMethod: Unknown method: maxDPI is not a valid method of interface razer.device.dpi

Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 880, in on_macroShortcutDialogOk_clicked
    if set_shortcut_stack.get_visible_child_name() == 'Keystroke':
NameError: name 'set_shortcut_stack' is not defined
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 880, in on_macroShortcutDialogOk_clicked
    if set_shortcut_stack.get_visible_child_name() == 'Keystroke':
NameError: name 'set_shortcut_stack' is not defined
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 880, in on_macroShortcutDialogOk_clicked
    if set_shortcut_stack.get_visible_child_name() == 'Keystroke':
NameError: name 'set_shortcut_stack' is not defined
```

GabMus, 2017-11-30:
At this point I'm pretty sure there's some driver stack bug. Will talk to the team, wait for an update from me.

GabMus, 2017-11-30:
Confirmed driver stack bug, can you please give me the output of `lsusb`?

Also mentioning @terrycain just cause

GabMus, 2017-11-30:
As of [this commit](https://github.com/openrazer/openrazer/compare/6e00366af445...d1e683ac7cba) the dpi should work. Please update to the latest driver version (openrazer-git packages)

jambonmcyeah, 2017-11-30:
@terrycain
```
[jambon@jambonarch ~]$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 1532:0053 Razer USA, Ltd 
Bus 001 Device 006: ID 1bcf:2b8c Sunplus Innovation Technology Inc. 
Bus 001 Device 005: ID 0cf3:e301 Qualcomm Atheros Communications 
Bus 001 Device 004: ID 187c:0528 Alienware Corporation 
Bus 001 Device 003: ID 046d:c333 Logitech, Inc. 
Bus 001 Device 002: ID 04f9:0249 Brother Industries, Ltd 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

jambonmcyeah, 2017-11-30:
After updating everything to the git version, it stopped working again
```
[jambon@jambonarch razercommander-git]$ razercommander
Importing device logic, waiting for daemon
Device logic loaded, daemon is alive
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 245, in initDevices
    newdev = device.Device(dev)
  File "/usr/lib/python3.6/site-packages/razercommander/device.py", line 50, in __init__
    self.device)
  File "/usr/lib/python3.6/site-packages/razercommander/macro_logic.py", line 73, in make_device
    n_device = MacroDevice(mk_arr, device_uid, device)
  File "/usr/lib/python3.6/site-packages/razercommander/macro_logic.py", line 14, in __init__
    self.set_all_current_macros()
  File "/usr/lib/python3.6/site-packages/razercommander/macro_logic.py", line 32, in set_all_current_macros
    key.set_current_macro(self.device)
  File "/usr/lib/python3.6/site-packages/razercommander/macro_logic.py", line 50, in set_current_macro
    self._set_macro(device, self.macro)
  File "/usr/lib/python3.6/site-packages/razercommander/macro_logic.py", line 43, in _set_macro
    device.macro.del_macro(self.key)
  File "/usr/lib/python3.6/site-packages/openrazer/client/macro.py", line 68, in del_macro
    self._macro_dbus.deleteMacro(bind_key)
  File "/usr/lib/python3.6/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3.6/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Python.AttributeError: Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3.6/site-packages/openrazer_daemon/dbus_services/dbus_methods/macro.py", line 30, in delete_macro
    self.key_manager.dbus_delete_macro(macro_key)
AttributeError: 'RazerNagaChroma' object has no attribute 'key_manager'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 687, in do_command_line
    self.do_activate()
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 661, in do_activate
    self.refreshDevices()
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 302, in refreshDevices
    self.initDevices()
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 248, in initDevices
    print('Skipping device: {}'.format(device.name))
AttributeError: module 'razercommander.device' has no attribute 'name'
```
```
[jambon@jambonarch razercommander-git]$ pacman -Q | grep razer
openrazer-daemon-git 2.1.1.r18.gd1e683a-1
openrazer-driver-dkms-git 2.1.1.r18.gd1e683a-1
python-openrazer-git 2.1.1.r18.gd1e683a-1
razercommander-git 1.2.0.r12.gab9d8a3-1
```

terrycain, 2017-11-30:
Yeah think thats the one i fixed.

jambonmcyeah, 2017-12-09:
My mouse now doesn't show up in razercommander for some reason
```
Importing device logic, waiting for daemon
Device logic loaded, daemon is alive
Skipping device Razer Naga Chroma
no devices
no devices
no devices
[jambon@jambonarch ~]$ pacman -Q | grep razer
openrazer-daemon-git 2.1.1.r18.gd1e683a-1
openrazer-driver-dkms-git 2.1.1.r18.gd1e683a-1
python-openrazer-git 2.1.1.r18.gd1e683a-1
razercommander-git 1.2.0.r20.g8ccff85-1
```

GabMus, 2017-12-10:
With the last commit if a device gets skipped it also prints the exception, so first update to that.
Also, you're not on the latest driver stack. Naga related issues should be solved in the very latest git version:

```bash
pacman -Q | grep openrazer
openrazer-daemon-git 2.1.1.r20.g4177be1-1
openrazer-driver-dkms-git 2.1.1.r20.g4177be1-1
python-openrazer-git 2.1.1.r20.g4177be1-1
```

jambonmcyeah, 2017-12-10:
I updated to the latest version and it won't even open now
```
[jambon@jambonarch razercommander-git]$ razercommander
Importing device logic, waiting for daemon
Device logic loaded, daemon is alive
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 247, in initDevices
    newdev = device.Device(dev)
  File "/usr/lib/python3.6/site-packages/razercommander/device.py", line 50, in __init__
    self.device)
  File "/usr/lib/python3.6/site-packages/razercommander/macro_logic.py", line 73, in make_device
    n_device = MacroDevice(mk_arr, device_uid, device)
  File "/usr/lib/python3.6/site-packages/razercommander/macro_logic.py", line 14, in __init__
    self.set_all_current_macros()
  File "/usr/lib/python3.6/site-packages/razercommander/macro_logic.py", line 32, in set_all_current_macros
    key.set_current_macro(self.device)
  File "/usr/lib/python3.6/site-packages/razercommander/macro_logic.py", line 50, in set_current_macro
    self._set_macro(device, self.macro)
  File "/usr/lib/python3.6/site-packages/razercommander/macro_logic.py", line 43, in _set_macro
    device.macro.del_macro(self.key)
  File "/usr/lib/python3.6/site-packages/openrazer/client/macro.py", line 68, in del_macro
    self._macro_dbus.deleteMacro(bind_key)
  File "/usr/lib/python3.6/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3.6/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Python.AttributeError: Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3.6/site-packages/openrazer_daemon/dbus_services/dbus_methods/macro.py", line 30, in delete_macro
    self.key_manager.dbus_delete_macro(macro_key)
AttributeError: 'RazerNagaChroma' object has no attribute 'key_manager'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 690, in do_command_line
    self.do_activate()
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 663, in do_activate
    self.refreshDevices()
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 304, in refreshDevices
    self.initDevices()
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 250, in initDevices
    print('Skipping device {}\n  Exception: {}'.format(dev.name), ex)
IndexError: tuple index out of range
```
```
[jambon@jambonarch razercommander-git]$ pacman -Q | grep razer
openrazer-daemon-git 2.1.1.r21.g71f52c8-1
openrazer-driver-dkms-git 2.1.1.r21.g71f52c8-1
python-openrazer-git 2.1.1.r21.g71f52c8-1
razercommander-git 1.2.1.r1.g2216127-1
```

GabMus, 2017-12-10:
@z3ntu any ideas?
by the way, should open now and just skip the naga.

z3ntu, 2017-12-10:
Yeah the keymanager was recently deactivated in the daemon (see https://github.com/openrazer/openrazer/commit/d1e683ac7cbaf0a70c6be1efc33d142f6c26ac72)

jambonmcyeah, 2018-01-06:
Any updates on this?