GabMus/razerCommander #45: Wont "open"

jjgrady12, 2017-09-19:
First shot, loads then errors out something about pyhton 3.5 - updated to python 3.6, even though python3 command shows 3.5.2 - reboot, now no message, just appears to load then disappears.

Ubuntu 16.04

Thanks!

jjgrady12, 2017-09-20:
josh@josh-Linux:~$ razercommander
Traceback (most recent call last):
  File "/usr/share/razercommander/main.py", line 7, in <module>
    import device
  File "/usr/share/razercommander/device.py", line 8, in <module>
    import razer.client as rclient
ImportError: No module named 'razer'
josh@josh-Linux:~$

hobarrera, 2017-09-20:
I'd suggest you surround your console outputs in triple-backticks to avoid weird formatting getting in the way:

    ```
    $ echo hi
    hi
    ```

hobarrera, 2017-09-20:
Can you please run `head $(which razercommander)`, and also `which python` and `python --version`.

Did you install the razer drivers package as well?

GabMus, 2017-11-28:
Any news on this?

Bifidus, 2017-11-30:
Hi

I have a similar problem (razercommander crash at startup). I did the commands WhyNotHugo suggested ::

b@moribond:~$ razercommander 
Traceback (most recent call last):
  File "/usr/bin/razercommander", line 44, in <module>
    gi.require_versions({
AttributeError: module 'gi' has no attribute 'require_versions'
b@moribond:~$ head $(which razercommander)
#!/usr/bin/python3

# razercommander
#
# Copyright (C) 2016 Gabriele Musco <emaildigabry@gmail.com>
#
# This program is free software: you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.
b@moribond:~$ which python
/usr/bin/python
b@moribond:~$ python --version
Python 2.7.12

Quax1102, 2017-12-09:
May I add some information here, as I have the same problem on Ubuntu 16.04 - razercommander starting, then showing the icon for some seconds and closing again.

First of all, I installed the driver package from OpenRazer, which should include the basic drivers neccessary (shouldn't it?).

I did the commands you suggested:
peter@PC-PETER:~$ head $(which razercommander)
#!/usr/bin/python3

# razercommander
#
# Copyright (C) 2016 Gabriele Musco <emaildigabry@gmail.com>
#
# This program is free software: you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.
peter@PC-PETER:~$ which python
/usr/bin/python
peter@PC-PETER:~$ python --version
Python 2.7.12

I then ran the install procedure for python 3.6, where the "install" part leaves me with the following information (messages translated from german):
"phyton 3.6 already is the newest version (3.6.3-1ubuntu1~16.04.york1)
The follwoing packages have been installed automatically and are no longer needed:
0ad-data 0ad-data-common fonts-play fonts-texgyre glade2script kamera
  kde-l10n-de kde-l10n-engb libenet7 libgloox13v5 libkf5itemmodels5
  libkf5kdcraw5 libkf5kipi30.0.0 libllvm3.8:i386 libnvtt2 libsdl2-2.0-0
  libsndio6.1 linux-headers-4.10.0-27 linux-headers-4.10.0-27-generic
  linux-headers-4.10.0-28 linux-headers-4.10.0-28-generic
  linux-headers-4.10.0-30 linux-headers-4.10.0-30-generic
  linux-headers-4.10.0-32 linux-headers-4.10.0-32-generic
  linux-headers-4.10.0-33 linux-headers-4.10.0-33-generic
  linux-headers-4.10.0-35 linux-headers-4.10.0-35-generic
  linux-headers-4.10.0-37 linux-headers-4.10.0-37-generic
  linux-headers-4.10.0-38 linux-headers-4.10.0-38-generic
  linux-headers-4.4.0-87 linux-headers-4.4.0-87-generic linux-headers-4.4.0-89
  linux-headers-4.4.0-89-generic linux-headers-4.8.0-36
  linux-headers-4.8.0-36-generic linux-image-4.10.0-27-generic
  linux-image-4.10.0-28-generic linux-image-4.10.0-30-generic
  linux-image-4.10.0-32-generic linux-image-4.10.0-33-generic
  linux-image-4.10.0-35-generic linux-image-4.10.0-37-generic
  linux-image-4.10.0-38-generic linux-image-4.4.0-87-generic
  linux-image-4.4.0-89-generic linux-image-4.8.0-36-generic
  linux-image-extra-4.10.0-27-generic linux-image-extra-4.10.0-28-generic
  linux-image-extra-4.10.0-30-generic linux-image-extra-4.10.0-32-generic
  linux-image-extra-4.10.0-33-generic linux-image-extra-4.10.0-35-generic
  linux-image-extra-4.10.0-37-generic linux-image-extra-4.10.0-38-generic
  linux-image-extra-4.4.0-87-generic linux-image-extra-4.4.0-89-generic
  linux-image-extra-4.8.0-36-generic pastebinit snap-confine tex-common
Use audo apt autoremove to remove them
0 actualized, 0 new installed, 0 removed and 142 not actualized

However, python --version still gives me 2.7.12

Any help would be appreciated.

GabMus, 2017-12-09:
@Quax1102 it should still use python3.
ubuntu uses python2 by default for python for some reason. if you launch razercommander it should explicitly use python3, not python2. you can check your python3 version like this: python3 --version

Bifidus, 2017-12-09:
Just an addon to my previous message, as i have a similar problem ::
b@moribond:~$ python3 --version
Python 3.5.2

GabMus, 2017-12-10:
You should [install python3.6 on ubuntu 16.04](https://askubuntu.com/questions/865554/how-do-i-install-python-3-6-using-apt-get/865569#865569). Anyway, I think that the best solution is updating to a more current ubuntu version like 17.10.

Bifidus, 2017-12-10:
I don't get it... i installed python 3.6, using the link you provide.
But the command still show python 3.5.2

(So far, i prefer to stick with LTS).

GabMus, 2017-12-11:
@Bifidus please send the output of `whereis python`

Bifidus, 2017-12-11:
```
b@moribond:~$ whereis python
python: /usr/bin/python3.5m /usr/bin/python2.7 /usr/bin/python /usr/bin/python3.6 /usr/bin/python3.5 /usr/bin/python3.6m /usr/lib/python2.7 /usr/lib/python3.6 /usr/lib/python3.5 /etc/python2.7 /etc/python /etc/python3.6 /etc/python3.5 /usr/local/lib/python2.7 /usr/local/lib/python3.6 /usr/local/lib/python3.5 /usr/include/python3.5m /usr/share/python /usr/share/man/man1/python.1.gz
```

GabMus, 2017-12-11:
@Bifidus Alright try to run /usr/bin/python3.6 /usr/bin/razercommander

Bifidus, 2017-12-11:
```
b@moribond:~$ /usr/bin/python3.6 /usr/bin/razercommander 
Traceback (most recent call last):
  File "/usr/bin/razercommander", line 42, in <module>
    import gi
  File "/usr/lib/python3/dist-packages/gi/__init__.py", line 42, in <module>
    from . import _gi
ImportError: cannot import name '_gi'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 63, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
  File "/usr/lib/python3/dist-packages/apport/__init__.py", line 5, in <module>
    from apport.report import Report
  File "/usr/lib/python3/dist-packages/apport/report.py", line 30, in <module>
    import apport.fileutils
  File "/usr/lib/python3/dist-packages/apport/fileutils.py", line 23, in <module>
    from apport.packaging_impl import impl as packaging
  File "/usr/lib/python3/dist-packages/apport/packaging_impl.py", line 23, in <module>
    import apt
  File "/usr/lib/python3/dist-packages/apt/__init__.py", line 23, in <module>
    import apt_pkg
ModuleNotFoundError: No module named 'apt_pkg'

Original exception was:
Traceback (most recent call last):
  File "/usr/bin/razercommander", line 42, in <module>
    import gi
  File "/usr/lib/python3/dist-packages/gi/__init__.py", line 42, in <module>
    from . import _gi
ImportError: cannot import name '_gi'
```

GabMus, 2017-12-11:
If I have to guess you need to install python-gobject for python 3.6.
Unfortunately dealing with mixed python versions is a mess... that's why I suggested you to upgrade.

Bifidus, 2017-12-11:
Fresh install of Xubuntu 17.10.
razercommander lauch, but don't see any device.
Here is the output on the console ::
```
b@moribond:~$ razercommander 
Importing device logic, waiting for daemon
Device logic loaded, daemon is alive
Skipping device Razer Tartarus Chroma
  Exception: Key 10 is not in openrazer.keyboard.KEY_MAPPING
no devices
no devices
no devices
```

GabMus, 2017-12-11:
great, it works! the issue you're experiencing is a completely different one, and it's driver related: #43 
you should try with the git version of the driver, not sure how to get it on ubuntu tho. probably your best bet is to build it from source.

Bifidus, 2017-12-11:
Ok. This is way above my knowledge.
I tried with the daily PPA, but the error is the same. I'll wait for there, i think. The daily PPA should be updated sonner than the release ppa, and i'll switch on the release one once everything work.
Thank you.

DanyGee, 2018-04-18:
Can anyone help me with this one?

razercommander
Importing device logic, waiting for daemon
ERROR: the daemon is not responding!
Try running `killall razer-daemon && razer-daemon` or rebooting. If this doesn't work, please fill an issue!
Exception: org.freedesktop.DBus.Python.FileNotFoundError: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/all.py", line 85, in get_keyboard_layout
    with open(driver_path, 'r') as driver_file:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/pci0000:00/0000:00:1c.4/0000:0d:00.0/usb5/5-2/5-2:1.2/0003:1532:021A.0006/kbd_layout'