GabMus/razerCommander #56: Logo wont change colors, deathadder chroma, arch

bcmmbc122333, 2017-10-31:
For some reason the color wont change for the logo on Arch. Im using a deathadder chroma. the color works on the scroll wheel but the logo doesnt. The effects work for both the scroll wheel and the logo. The only time the logo shows another color is during the spectrum cycle effect, it will start red and fade into to black then fade to blue, then blue will fade into red again and the cycle repeats.

GabMus, 2017-11-28:
Sounds like the green part of your rgb led is fried. If you can, try the windows driver, if it doesn't work either, RMA it

GabMus, 2017-12-10:
closed for inactivity, also this is probably hardware related. if I'm wrong, just reopen it

closed 2017-12-10 by GabMus