GabMus/razerCommander #57: Manjaro 17.0.6 Gellivara - Razer Naga Chroma

ShaolinWarrior, 2017-11-07:
Still new to all of this, so please forgive me if I'm in the wrong spot, or if I'm posting the wrong information. I tried to do as much troubleshooting on my own, and, it may be that I have done everything right, but that the issue is elsewhere unrelated to this project. I'll let you be the judge of that, and hopefully we can get my issue resolved together.

Issue: Installed (what feels like everything), however I can't open the RazerCommander program to modify the side buttons on my Razer Naga Chroma. According to [OpenRazer](https://openrazer.github.io/) and [RazerCommander](https://gabmus.github.io/razerCommander/), after I have installed everything, I should be able to configure my mouse - right? Maybe? Polychromatic and Razergenie both seem to be recognizing my device without issue.
When I say it 'won't open, I mean exactly that. I click the icon, and it acts like it's going to open for just a moment, then it goes away. Below is my system details, installs regarding this, and logs gathered up from other people's questions - hopefully they help.

My setup:

Manjaro 17.0.6 Gellivara
x86_64 Linux 4.9.60-1-MANJARO
KDE 5.39.0 / Plasma 5.11.2
Razer Naga Chroma

Installed headers using:
`pacman -S linux49-headers`

plugdev:
`sudo gpasswd -a meihua plugdev`

Rebooted

Installed:
Polychromatic
Razercommander-git
Razergenie-git

rebooted

Output of:
sudo dkms install razer_chroma_driver/1.0.0
`Module razer_chroma_driver/1.0.0 already installed on kernel 4.9.60-1-MANJARO/x86_64`

Output of:
python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'razer'
```

Output of:
sudo modprobe razermouse && lsmod | grep razer
```
razermouse             65536  0
hid                   114688  5 i2c_hid,razermouse,hid_generic,usbhid,hid_multitouch
usbcore               208896  9 razermouse,uvcvideo,usbhid,ehci_hcd,xhci_pci,rtsx_usb,btusb,xhci_hcd,ehci_pci
```

Output of:
sudo udevadm test /sys/bus/hid/devices/0*:1532:0053.*
```
calling: test
version 235
This program is for debugging only, it does not run any program
specified by a RUN key. It may show incorrect results, because
some values may be different, or not available at a simulation run.

=== trie on-disk ===
tool version:          235
file size:         8988132 bytes
header size             80 bytes
strings            1930108 bytes
nodes              7057944 bytes
Load module index
Found container virtualization none.
timestamp of '/etc/systemd/network' changed
timestamp of '/usr/lib/systemd/network' changed
Ignoring /usr/lib/systemd/network/80-container-host0.network, because it's not a regular file with suffix .link.
Ignoring /usr/lib/systemd/network/80-container-vz.network, because it's not a regular file with suffix .link.
Ignoring /usr/lib/systemd/network/80-container-ve.network, because it's not a regular file with suffix .link.
Parsed configuration file /usr/lib/systemd/network/99-default.link
[... 1199 lines cut ...]
Nov 07 04:25:38 FuguKDE razer_mount[6135]: Driver razermouse
Nov 07 04:25:38 FuguKDE razer_mount[6136]: Device_ID 0003:1532:0053.0009
Nov 07 04:25:38 FuguKDE razer_mount[6137]: Unbinding 0003:1532:0053.000A from hid-generic
Nov 07 04:25:38 FuguKDE razer_mount[6139]: Unbinding 0003:1532:0053.0009 from hid-generic
Nov 07 04:25:38 FuguKDE razer_mount[6142]: Driver razermouse
Nov 07 04:25:38 FuguKDE razer_mount[6143]: Device_ID 0003:1532:0053.000B
Nov 07 04:25:38 FuguKDE razer_mount[6144]: Unbinding 0003:1532:0053.000B from hid-generic
Nov 07 04:25:38 FuguKDE razer_mount[6145]: Binding 0003:1532:0053.000A to razermouse
Nov 07 04:25:38 FuguKDE razer_mount[6146]: Binding 0003:1532:0053.0009 to razermouse
Nov 07 04:25:38 FuguKDE razer_mount[6570]: Binding 0003:1532:0053.000B to razermouse
Nov 07 04:25:38 FuguKDE razer_mount[6659]: Finished binding 0003:1532:0053.000A
Nov 07 04:25:38 FuguKDE razer_mount[6660]: Finished binding 0003:1532:0053.0009
Nov 07 04:25:38 FuguKDE razer_mount[6661]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0053.000A/
Nov 07 04:25:38 FuguKDE razer_mount[6662]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0053.0009/
Nov 07 04:25:38 FuguKDE razer_mount[6665]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0053.000A/
Nov 07 04:25:38 FuguKDE razer_mount[6667]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0053.0009/
Nov 07 04:25:38 FuguKDE razer_mount[6672]: Finished binding 0003:1532:0053.000B
Nov 07 04:25:38 FuguKDE razer_mount[6673]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0053.000B/
Nov 07 04:25:38 FuguKDE razer_mount[6675]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0053.000B/
```

Output of:
sudo ~/.local/share/openrazer/logs/razer.log

`sudo: /home/meihua/.local/share/openrazer/logs/razer.log: command not found`

ShaolinWarrior, 2017-11-11:
Octopi was telling me I had some updates available so I ran 
`pacaur -S`

After the installation of the AUR updates, it gave me a little advice and reccomended I run the following:
`systemctl --user enable --now openrazer-daemon.service`

Seeing as it had to do with the Razer daemon service, I figured that was wise advice and ran it.

I then ran into an error saying that a DKMS module wasn't installed, specifically -
`4.9.47-1-rt37-MANJARO`
(yes module, not headers)

So I installed with
`sudo dkms autoinstall -k 4.9.47-1-rt37-MANJARO`

Which installed without issue and without any further tips or advice from the system.

Now when I run
`λ ~/ dkms status`

I get the following output:
```
razer_chroma_driver, 1.0.0, 4.9.47-1-rt37-MANJARO, x86_64: installed
razer_chroma_driver, 1.0.0, 4.9.Tried running it verbose to see if that would give a hint to the problem, here's the output:60-1-MANJARO, x86_64: installed
```
So I'm thinking sweet, maybe that was the one thing I was missing, so I tried running RazerCommander, and - No luck.

So I tried running it again from the terminal with the -v argument to see a verbose log, here's the output:

```
λ ~/ razercommander -v      
Traceback (most recent call last):
  File "/usr/bin/razercommander", line 52, in <module>
    from razercommander import __main__
  File "/usr/lib/python3.6/site-packages/razercommander/__main__.py", line 250
    print('Skipping device: %s', % device.name)
```

I don't understand the output myself at this moment in time, but I will keep digging into it.

GabMus, 2017-11-25:
try again later with the latest commit

GabMus, 2017-12-10:
this is probably a duplicate of #43 
closing, if it's not a duplicate, please reopen it.

closed 2017-12-10 by GabMus