GabMus/razerCommander #58: AttributeError: model 'gi'

Kauri5, 2017-11-10:
I installed version 1.2.0. When I start it from the menu, nothing happens.

When I start it from the command line, I get the following error message:

```
Traceback (most recent call last):
  File "/usr/bin/razercommander", line 44, in <module>
    gi.require_versions({
AttributeError: module 'gi' has no attribute 'require_versions'
```

What can I do?

Xcelled, 2017-11-16:
I fixed this by editing `/usr/bin/razercommander`.

I changed the `gi.require_versions` lines to `gi.require_version('Gtk', '3.0')` and it worked after that.

Kauri5, 2017-11-18:
Now it works. Thanks a lot!

Xcelled, 2017-11-18:
@Kauri5 this is an actual issue. You should leave it open so it can be fixed.

GabMus, 2017-11-24:
should work now

closed 2017-11-24 by GabMus