GabMus/razerCommander #60: Cannot Create Keyboard Macro

vozeldr, 2017-11-22:
This is when trying to create a keyboard macro on my Blackwidow Chroma.

Traceback (most recent call last):
  File "/home/xxxxx/Projects/razerCommander/razercommander/__main__.py", line 882, in on_macroShortcutDialogOk_clicked
    if set_shortcut_stack.get_visible_child_name() == 'Keystroke':
NameError: name 'set_shortcut_stack' is not defined

GabMus, 2017-11-25:
What version of razercommander do you have? main.py in the root of the git repo doesn't exist anymore

vozeldr, 2017-11-29:
I cloned the latest git repo and followed the instructions here to build and run it...
https://github.com/GabMus/razerCommander#building-for-testing

GabMus, 2017-11-30:
Sounds unlikely... Please try cloning again and compiling in a new folder.
Also, what distro are you on?

vozeldr, 2017-11-30:
I'm on Ubuntu 17.10 upgraded from 17.04. Here is the output from the console of a fresh clone, build and run (all I did is find/replace my computer and user name).

uname@cname:~/.config$ tar -czvf razercommander.tgz razercommander
razercommander/
razercommander/macro_logic.json
razercommander/profiles
uname@cname:~/.config$ rm -rf razercommander
uname@cname:~/.config$ cd openrazer
uname@cname:~/.config/openrazer$ ls
razer.conf
uname@cname:~/.config/openrazer$ cd ../
uname@cname:~/.config$ mv razercommander.tgz /data
uname@cname:~/.config$ cd ../Projects/
uname@cname:~/Projects$ git clone git@github.com:GabMus/razerCommander.gitCloning into 'razerCommander'...
remote: Counting objects: 1028, done.
remote: Compressing objects: 100% (17/17), done.
remote: Total 1028 (delta 7), reused 13 (delta 4), pack-reused 1007
Receiving objects: 100% (1028/1028), 9.93 MiB | 2.30 MiB/s, done.
Resolving deltas: 100% (485/485), done.
uname@cname:~/Projects$ cd razerCommander/
uname@cname:~/Projects/razerCommander$ mkdir builddir
uname@cname:~/Projects/razerCommander$ cd builddir
uname@cname:~/Projects/razerCommander/builddir$ meson ..
The Meson build system
Version: 0.42.1
Source dir: /home/uname/Projects/razerCommander
Build dir: /home/uname/Projects/razerCommander/builddir
Build type: native build
Project name: razercommander
Build machine cpu family: x86_64
Build machine cpu: x86_64
Configuring org.gabmus.razercommander.desktop using configuration
Configuring org.gabmus.razercommander.service using configuration
Found pkg-config: /usr/bin/pkg-config (0.29.1)
Native dependency glib-2.0 found: YES 2.54.1
Program glib-compile-schemas found: YES (/usr/bin/glib-compile-schemas)
Configuring razercommander using configuration
Program scripts/meson_post_install.py found: YES (/usr/bin/env python3 /home/uname/Projects/razerCommander/scripts/meson_post_install.py)
Build targets in project: 3
uname@cname:~/Projects/razerCommander/builddir$ meson configure -Dprefix=$(pwd)/testdir
uname@cname:~/Projects/razerCommander/builddir$ ninja install
[0/1] Regenerating build files.
The Meson build system
Version: 0.42.1
Source dir: /home/uname/Projects/razerCommander
Build dir: /home/uname/Projects/razerCommander/builddir
Build type: native build
Project name: razercommander
Build machine cpu family: x86_64
Build machine cpu: x86_64
Configuring org.gabmus.razercommander.desktop using configuration
Configuring org.gabmus.razercommander.service using configuration
Found pkg-config: /usr/bin/pkg-config (0.29.1)
Native dependency glib-2.0 found: YES 2.54.1
Program glib-compile-schemas found: YES (/usr/bin/glib-compile-schemas)
Configuring razercommander using configuration
Program scripts/meson_post_install.py found: YES (/usr/bin/env python3 /home/uname/Projects/razerCommander/scripts/meson_post_install.py)
Build targets in project: 3
[1/2] Generating org.gabmus.razercommander_gresource with a custom command.
XMLLINT not set and xmllint not found in path; skipping xml preprocessing.
[1/2] Installing files.
Installing subdir /home/uname/Projects/razerCommander/razercommander to /home/uname/Projects/razerCommander/builddir/testdir/lib/python3.6/site-packages
Installing data/org.gabmus.razercommander.gresource to /home/uname/Projects/razerCommander/builddir/testdir/share/razercommander/org.gabmus.razercommander.gresource
Installing /home/uname/Projects/razerCommander/builddir/data/org.gabmus.razercommander.desktop to /home/uname/Projects/razerCommander/builddir/testdir/share/applications
Installing /home/uname/Projects/razerCommander/builddir/data/org.gabmus.razercommander.service to /home/uname/Projects/razerCommander/builddir/testdir/share/dbus-1/services
Installing /home/uname/Projects/razerCommander/data/icons/org.gabmus.razercommander.svg to /home/uname/Projects/razerCommander/builddir/testdir/share/icons/hicolor/scalable/apps
Installing /home/uname/Projects/razerCommander/builddir/bin/razercommander to /home/uname/Projects/razerCommander/builddir/testdir/bin
Running custom install script '/usr/bin/env python3 /home/uname/Projects/razerCommander/scripts/meson_post_install.py'
Compiling gsettings schemas...
Error opening directory “/home/uname/Projects/razerCommander/builddir/testdir/share/glib-2.0/schemas”: No such file or directory
[... 24 lines cut ...]
    n_device = MacroDevice(mk_arr, device_uid, device)
  File "/home/uname/Projects/razerCommander/razercommander/macro_logic.py", line 14, in __init__
    self.set_all_current_macros()
  File "/home/uname/Projects/razerCommander/razercommander/macro_logic.py", line 32, in set_all_current_macros
    key.set_current_macro(self.device)
  File "/home/uname/Projects/razerCommander/razercommander/macro_logic.py", line 50, in set_current_macro
    self._set_macro(device, self.macro)
  File "/home/uname/Projects/razerCommander/razercommander/macro_logic.py", line 43, in _set_macro
    device.macro.del_macro(self.key)
  File "/usr/lib/python3/dist-packages/openrazer/client/macro.py", line 66, in del_macro
    raise ValueError("Key {0} is not in openrazer.keyboard.KEY_MAPPING".format(bind_key))
ValueError: Key 10 is not in openrazer.keyboard.KEY_MAPPING

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/uname/Projects/razerCommander/razercommander/__main__.py", line 687, in do_command_line
    self.do_activate()
  File "/home/uname/Projects/razerCommander/razercommander/__main__.py", line 661, in do_activate
    self.refreshDevices()
  File "/home/uname/Projects/razerCommander/razercommander/__main__.py", line 302, in refreshDevices
    self.initDevices()
  File "/home/uname/Projects/razerCommander/razercommander/__main__.py", line 248, in initDevices
    print('Skipping device: {}'.format(device.name))
AttributeError: module 'razercommander.device' has no attribute 'name'

vozeldr, 2017-11-30:
Also, my python version is Python 3.6.3 and openrazer version 2.1.1~ubuntu17.10.1

[image]

vozeldr, 2017-11-30:
In order to get past this error and run the program (I submitted this fix in another issue), on line 248 of __main__.py change device.name to device.device.name

Afterwards, I can successfully run the program and I see the following in the console.

```
[0/1] Running external command run.
Importing device logic, waiting for daemon
Device logic loaded, daemon is alive
Skipping device: Razer Kraken 7.1 Chroma (Rainie)
```

The problem with that is the Razer Kraken 7.1 Chroma (Rainie) device shows up in the the dropdown in razerCommander... it's the Tartarus that *should* be shown as being skipped.

vozeldr, 2017-11-30:
So after that, back to the original issue...

```
[0/1] Running external command run.
Importing device logic, waiting for daemon
Device logic loaded, daemon is alive
Skipping device: Razer Kraken 7.1 Chroma (Rainie)
Traceback (most recent call last):
  File "/home/uname/Projects/razerCommander/razercommander/__main__.py", line 880, in on_macroShortcutDialogOk_clicked
    if set_shortcut_stack.get_visible_child_name() == 'Keystroke':
NameError: name 'set_shortcut_stack' is not defined
```
That happens after choosing my Razer BlackWidow Chroma, hitting Macros, clicking Unassigned to bring up the dialog. The same error message displays when I hit the OK button whether I try recording a keystroke or entering an xdotool command manually.

vozeldr, 2017-11-30:
OK, I fixed the issue locally by adding self in front of set_shortcut_stack on line 880 of `__main__.py`... so the TLDR from all of this is to make these 2 changes (although something is still wrong with the device name that actually gets printed).

```
Line 248: print('Skipping device: {}'.format(device.device.name))
Line 880: if self.set_shortcut_stack.get_visible_child_name() == 'Keystroke':
```

GabMus, 2017-12-09:
thanks for the patch, I applied it and now it should work. please test it out and let me know.

vozeldr, 2017-12-11:
It appears to be working fine now, thanks!

closed 2017-12-11 by vozeldr