GabMus/razerCommander #66: i need help my divice isnt showing in razercommander

zachary12345678, 2017-12-19:
my device isnt showing in razer commander i got the application and it says no devices connected and i have my razer ornata plugged in.

GabMus, 2017-12-19:
What distro are you running? What version of the driver did you install? How did you install razercommander?

Answer these questions, then open up a terminal, write `razercommander`, press enter and send the output.

Kah-leeb, 2017-12-19:
OK so i downloaded this on my Ubuntu then i downloaded these 3 links and one of them i clicked on them and it opened an installer for razerCommander then I installed it and I install python 3.6 also

zachary12345678, 2017-12-19:
OK so i downloaded this on my Ubuntu https://openrazer.github.io/#ubuntu then i downloaded these 3 links https://github.com/GabMus/razerCommander/releases/tag/1.2.1.1 and one of them i clicked on them and it opened an installer for razerCommander then I installed it and I install python 3.6 also and thats all i did

Kah-leeb, 2017-12-19:
hey i did the same thing and it didnt work

GabMus, 2017-12-19:
@zachary12345678 did you reboot after installing everything?

zachary12345678, 2017-12-19:
let me try that

zachary12345678, 2017-12-19:
nope nothing has changed

GabMus, 2017-12-20:
> how do i reboot

Power off your computer and turn it back on.

> did i install everything i needed?

If you installed openrazer-drivers, openrazer-daemon and python3-openrazer, yes you did.

> I installed using the razercommander_1.2.1.1ubuntu1_all.deb file

That's correct.

zachary12345678, 2017-12-20:
So why isn't my razer ornata linking to razerCommander I also did pollychromatic I think that's what's it's called and the same message popped up which was no devices found and I had my razer ornata plugged in

GabMus, 2017-12-20:
Again, did you reboot after installing the drivers?

Kah-leeb, 2017-12-20:
Yes

zachary12345678, 2017-12-20:
Yes I have but who's Kah-leeb and why does he keep on randomly join this conversion

GabMus, 2017-12-20:
He's another user that apparently has the same problem as you.

Anyway, run this command and send over the output: `lsmod | grep razer`

GabMus, 2017-12-20:
Also @zachary12345678 try using *punctuation* in your comments. Trying to understand any of your comments is a real challenge .

zachary12345678, 2017-12-20:
OK, I will.

zachary12345678, 2017-12-20:
OK, I ran the command in Xterm and nothing happened.

GabMus, 2017-12-20:
Good, this means that you don't have the kernel driver installed. The reason is that you probably don't have the headers for your current kernel, try running this command: `sudo apt install linux-generic linux-headers-generic`.

This command will ask for your password. When you're done, reboot your computer and try opening up razercommander again. If it still doesn't work, do the following:

- run `razercommander` from the terminal again, and post the output here.
- run `lsmod | grep razer` again and post the output here.

zachary12345678, 2017-12-20:
I ran the commands and downloaded the stuff you told me and nothing has changed it still says no devices connected. Then when I open the terming and run razercommander it opens up and it says 3 or 4 times no devices. Then I type in lsmod | grep razer nothing happens stuff it still says No devices Connected on razercommander.

GabMus, 2017-12-20:
> I ran the commands and downloaded the stuff you told me

did you reboot after running the command?

zachary12345678, 2017-12-20:
Yes, I rebooted and then I typed in razercommander then I typed in lsmod | grep razer

GabMus, 2017-12-21:
Run the following and post the output: `dpkg --list | grep razer`

Also, how did you install the openrazer driver? Using the ppa?

zachary12345678, 2017-12-21:
When I type in dpkg --list | grep razer in the Xterm it says ii  openrazer-daemon                      2.1.1~ubuntu16.04.1                        amd64        OpenRazer Device Daemon
ii  openrazer-doc                         2.1.1~ubuntu16.04.1                        all          Documentation and examples for the OpenRazer Tools
ii  openrazer-kernel-modules-dkms         2.1.1~ubuntu16.04.1                        amd64        OpenRazer Peripheral Drivers
ii  openrazer-meta                        2.1.1~ubuntu16.04.1                        all          Metapackage for OpenRazer components
ii  python3-openrazer                     2.1.1~ubuntu16.04.1                        all          OpenRazer Python Library
ii  razercommander                        1.2.1.1ubuntu1                             all          Razer device manager for Linux

zachary12345678, 2017-12-21:
When I run dpkg --list | grep razer it says 
ii  openrazer-daemon                      2.1.1~ubuntu16.04.1
     amd64        OpenRazer Device Daemon
ii  openrazer-doc                         2.1.1~ubuntu16.04.1
    all          Documentation and examples for the OpenRazer Tools
ii  openrazer-kernel-modules-dkms         2.1.1~ubuntu16.04.1
    amd64        OpenRazer Peripheral Drivers
ii  openrazer-meta                        2.1.1~ubuntu16.04.1
     all          Metapackage for OpenRazer components
ii  python3-openrazer                     2.1.1~ubuntu16.04.1
     all          OpenRazer Python Library
ii  razercommander                        1.2.1.1ubuntu1
      all          Razer device manager for Linux

zachary12345678, 2017-12-21:
I'm sorry I don't know why some of the XTerm code is crossed out

zachary12345678, 2017-12-21:
And also when I type in sudo modprobe razerkbd it asks for my password and then it says not found in directory /lib/modules/3.8.11

GabMus, 2017-12-21:
@zachary12345678 Use gnome-terminal instead of xterm and post the full output of those commands

zachary12345678, 2017-12-21:
When I type gnome-terminal in Xterm it says command not found

GabMus, 2017-12-22:
Are you sure you are on Ubuntu 16.04?

Post the output of `lsb_release -a`

zachary12345678, 2017-12-22:
This is what it says when I run the command lsb_release -a 
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.3 LTS
Release:        16.04
Codename:       xenial

zachary12345678, 2017-12-26:
hello?

GabMus, 2018-01-10:
This looks like either an upstream issue or something wrong you did that I can't see myself. Either way, it's not a razerCommander specific issue, I think you'll have better luck opening an issue on the upstream driver repo: https://github.com/openrazer/openrazer/issues/new

closed 2018-01-10 by GabMus