GabMus/razerCommander #70: Sync razer devices

JeromeDesseaux, 2018-02-28:
Hi and thanks for all the work you've done! 

I found a missing use case that I used often on windows : Device color sync. The chroma color of the mouse is the same at any time than the keyboard's one. 

Have you planned such fonctionnality? 

Thanks :)

GabMus, 2018-03-02:
Should be there already, it's controlled by the Sync switch in the headerbar.

It doens't always work tho, and that's weird, can't tell what's the problem, if it's in my frontend or in the driver stack itself, will need to look deeper into it.

JeromeDesseaux, 2018-03-05:
I updated razercommander and, in fact, the sync toggle button is there. 

Anyway, it does work on current session. But on recovery from suspend (or even reboot), sync is lost. 

I think it's an openrazer issue. But I'd like to ask you before issuing this on openrazer's github page.

Thanks for the response :)